esphome/esphome-core #3: Topics

fabaff, 2018-01-23:
The topic is derived from `app.set_name("xyz")`. Wouldn't it make sense to allow to set a prefix for the device topics?

I guess that I'm not the only one that groups sensors/actuators per location, e.g. `home/living_room`, to be able to subscribe to a room with wildcards.

OttoWinter, 2018-01-23:
Hmmm, haven't thought of that before - all the people I've shared this with privately group all devices per node. But I think you should be able to do something like `app.set_name("home/living_room")` because esphomelib doesn't really care much about your name.

I will check this later today because maybe the WiFi hostname generation fails with such a name. But then one could also manually override the hostname on the `WiFiComponent` with `set_hostname()`.

OttoWinter, 2018-01-23:
Oh, the name is included in the discovery topic (`discovery_prefix`/`component_type`/`name`/`entity_id`/config`) - so the "hack" would probably break discovery.

The thing is, I want to keep this library as simple as possible to use, but you're right, this functionality would probably be quite useful. Maybe a potential fix for this would be to have a separate MQTT prefix, that would by default be set to the name, but can be overridden. And then have some whitelisted chars for the discovery topic name. Do you have any suggestions for this?

fabaff, 2018-01-24:
I agree that things should be simple. Playing with `app.set_name()` was my first step just to make the same observation with the discovery topic as you did :-)

Unfortunately I have no real suggestion about the topics because almost everyone has it's own style. At some point in time the discovery topic in HA was extended to cover more ground (not only `object_id` but now we have also `node_id`). We simple use a regex to do some checking.

From my point of view is some flexibility needed because in a lot of cases you want to integrate a device into an existing environment and people will turn away if it's too limited. At the end of the day it's a simple trade-off between simplicity and usability.

OttoWinter, 2018-01-24:
Yes, you're completely right - and supporting these edge case (or in this case rather useful cases) is  probably what really makes a library great . So I just added this functionality on the master branch and pushed an update. The way this can be done now is like this: First, set your name to anything you like (for example `app.set_name("xyz");`, then do the following:

```cpp
auto *mqtt = app.init_mqtt(...);
mqtt->set_topic_prefix("home/living_room");
```

Thank you for reporting this!

closed 2018-01-24 by OttoWinter