esphome/esphome-core #5: Fans

fabaff, 2018-01-24:
The code for fans is available but it seems that `app` is not aware of that.

`esphomelib` seems to be the first tool that would allow one to control fan with an ESP.

OttoWinter, 2018-01-24:
Oh, didn't notice that - I have so many branches here locally and I probably forgot to include that - sorry for that. Thank you!

Until an update is pushed you can do the following (essentially just what `Application` do internally)

```cpp
auto *fan_state = new fan::FanState();
auto *mqtt_fan = new fan::MQTTFanComponent("Livingroom Radiator Fan");
mqtt_fan->set_state(fan_state);
auto *fan = new fan::BasicFanComponent();
fan->set_state(fan_state);

// use any output component you like
auto *fan_float = app.make_ledc_component(22, 50000);
// 0.0 -> off_speed
// 0.25 -> low_speed
// 0.6 -> medium_speed
// 1.0 -> high_speed 
fan->set_speed(fan_float, 0.0, 0.25, 0.6, 1.0);
app.register_mqtt_component(mqtt_fan);
app.register_component(fan);
```

fabaff, 2018-01-24:
Cool, thanks.

OttoWinter, 2018-01-25:
Fixed with version v0.3.0, but I'm still waiting for platformio to register the update. Thanks for reporting this!

closed 2018-01-25 by OttoWinter