esphome/esphome-core #6: lroundf is not a member of std

Big4SMK, 2018-01-25:
When trying to compile a nearly empty project that is using your library I get compile errors like:

.piolibdeps\esphomelib_ID2096\src\esphomelib\sensor\mqtt_sensor_component.cpp: In member function 'void esphomelib::sensor::MQTTSensorComponent::push_out_value(float, int8_t)':
.piolibdeps\esphomelib_ID2096\src\esphomelib\sensor\mqtt_sensor_component.cpp:63:25: error: 'lroundf' is not a member of 'std'
float value_rounded = std::lroundf(value * multiplier) / multiplier;

Any hints on how to solve this?

OttoWinter, 2018-01-25:
Hmmm, apparently your c++ standard library doesn't have the `lroundf` method (should be in `math.h`). Anyway, could you please try running `platformio update` and `platformio upgrade` to update the toolchain and give some information about what system this is appearing on (windows, unix...) and the output of `platformio --version`? Thank you!

Big4SMK, 2018-01-25:
I am running on windows, but I can't seem to find what my stdlib version is (I am rather new to platformio)
From what I've read it should be supported in at least visual studio 2013. 

> Big4SMK@MyOwnDesktop MINGW64 /j/Programming/MyESP32
> $ platformio update
> Updating tool-scons                      @ 3.20501.2      [Up-to-date]
> Updating tool-unity                      @ 1.20302.1      [Up-to-date]
> Updating contrib-pysite                  @ 0.1.4          [Up-to-date]
> Updating contrib-piohome                 @ 0.6.1          [Up-to-date]
> Updating tool-pioplus                    @ 0.13.4         [Up-to-date]
> 
> Platform Manager
> ================
> Platform Espressif 32
> --------
> Updating espressif32                     @ 0.11.0         [Up-to-date]
> Updating tool-esptoolpy                  @ 1.20100.0      [Up-to-date]
> Updating framework-arduinoespressif32    @ 1.3.2          [Up-to-date]
> 
> Platform Espressif 8266
> --------
> Updating espressif8266                   @ 1.6.0          [Up-to-date]
> Updating tool-esptool                    @ 1.409.0        [Up-to-date]
> Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
> 
> 
> Library Manager
> ===============
> Updating ArduinoJson                     @ 5.13.0         [Up-to-date]
> Updating DHT                             @ 1.0.0          [Up-to-date]
> Updating DallasTemperature               @ 3.7.8          [Up-to-date]
> Updating OneWire                         @ 2.3.2          [Up-to-date]
> Updating PCA9685 16-Channel PWM Driver Module Library @ 22b7212        [Up-to-date]
> Updating PubSubClient                    @ 2.6            [Up-to-date]
> Updating esphomelib                      @ 0.1.3          [Up-to-date]
> 
> Big4SMK@MyOwnDesktop MINGW64 /j/Programming/MyESP32
> $ platformio upgrade
> You're up-to-date!
> PlatformIO 3.5.2a3 is currently the newest version available.
> 
> Big4SMK@MyOwnDesktop MINGW64 /j/Programming/MyESP32
> $ platformio --version
> PlatformIO, version 3.5.2a3

OttoWinter, 2018-01-25:
Ok, thanks. I mean the function should clearly be there (https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/newlib/math.h#L349) - maybe it's some kind of namespace issue, I'm certainly no expert on this.

Anyway, could you try editing your `.piolibdeps\esphomelib_ID2096\src\esphomelib\sensor\mqtt_sensor_component.cpp` file and replace `lroundf` with `round` and report whether that worked? Thanks.

Big4SMK, 2018-01-25:
using round gives me another error
`.piolibdeps\esphomelib_ID2096\src\esphomelib\sensor\mqtt_sensor_component.cpp: In member function 'void esphomelib::sensor::MQTTSensorComponent::push_out_value(float, int8_t)':
C:\users\big4smk.myowndomain\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:78:22: error: expected unqualified-id before '(' token
#define round(x)     ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
^
.piolibdeps\esphomelib_ID2096\src\esphomelib\sensor\mqtt_sensor_component.cpp:63:30: note: in expansion of macro 'round'
float value_rounded = std::round(value * multiplier) / multiplier;
^`

but I also think I should have included a larger part of the log on my initial lroundf error, as it does actually seem to find the macro you are referring to:

`.piolibdeps\esphomelib_ID2096\src\esphomelib\sensor\mqtt_sensor_component.cpp: In member function 'void esphomelib::sensor::MQTTSensorComponent::push_out_value(float, int8_t)':
.piolibdeps\esphomelib_ID2096\src\esphomelib\sensor\mqtt_sensor_component.cpp:63:25: error: 'lroundf' is not a member of 'std'
float value_rounded = std::lroundf(value * multiplier) / multiplier;
^
.piolibdeps\esphomelib_ID2096\src\esphomelib\sensor\mqtt_sensor_component.cpp:63:25: note: suggested alternative:
In file included from C:\users\big4smk.myowndomain\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal.h:34:0,
from C:\users\big4smk.myowndomain\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:35,
from .piolibdeps\PubSubClient_ID89\src/PubSubClient.h:10,
from .piolibdeps\esphomelib_ID2096\src/esphomelib/mqtt/mqtt_client_component.h:14,
from .piolibdeps\esphomelib_ID2096\src/esphomelib/mqtt/mqtt_component.h:10,
from .piolibdeps\esphomelib_ID2096\src\esphomelib\sensor\mqtt_sensor_component.h:9,
from .piolibdeps\esphomelib_ID2096\src\esphomelib\sensor\mqtt_sensor_component.cpp:5:
C:\users\big4smk.myowndomain\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\newlib/math.h:349:17: note:   'lroundf'
extern long int lroundf _PARAMS((float));
^
*** [.pioenvs\test\libefc\esphomelib_ID2096\esphomelib\sensor\mqtt_sensor_component.o] Error 1`

OttoWinter, 2018-01-25:
Oh thanks. I would now guess MinGW (which I believe is the compiler you're using) handles the *standardized* C++ `#include<cmath>` differently from the gcc compiler - especially the `std::` namespace wrapping. But I guess the old C-way of using `math.h` should still work with MinGW.

Sorry that this happened. I don't have any windows PCs around that I tested this library on before publishing. I will try to update the library tomorrow or today to hopefully fix this.

Big4SMK, 2018-01-25:
That makes sense, but how would you suggest to fix this? I think this is actually a problem with the framework as that is depending on the lroundf function to be available in the lib. The only way to fix it in your code is to not use the lroundf macro at all, right?
If there is any idea you got while I am also trying some things, please let me know.

OttoWinter, 2018-01-25:
Ok, history time 😂  - as you might know, the C++ language, which this library is based on, is (more or less) an extension to the C language. Back in the day when only the C language existed, people used to `#include <math.h>` for all math operations and that worked just fine.

However, when C++ came along, it introduced `namespace`s, which can encapsulate methods so that, if two libraries have methods with the same name, you can still choose which one you would like to use. For example, if the library (or technically the header file) `Foo.h` has a method called `asdf()` inside a namespace `foo`and `Bar.h` has a method with the same name, but in the `bar` namespace - your program can choose *which of those methods* it wants to use by specifying the namespace - so if you'd like to call `Foo`'s `asdf()` you'd call `foo::asdf()`. This was not possible in C and was great for bigger C++ projects.

Next, to keep the global namespace clean, the C++ standard library tries to encapsulate all its helper methods inside the `std` namespace. But if you're including `math.h`, those helper methods are not inside the `std` namespace because it's legacy C code. To fix this and keep the standard library consistent in using the `std` namespace, the `#include <cmath>` include was introduced - this new header file basically just had all functions of `math.h`, but in the `std` namespace. So if you used to call `round()` using `math.h`, you would now have to call `std::round()` - the same applies for `lroundf()`. You're even encouraged by most compilers to use the new `cmath` include, because it's considered to be better C++ code - that's also why I used it. Anyway, MinGW apparently doesn't adhere to the standards, which to be fair is quite common in compiler land, and doesn't fully support `cmath` in the way I used it.

So to answer your question, *a* way to fix this would thereby be to just use the old C way of using the `math.h` include and that would probably not cause any issues. 

But there's another (tiny) issue: the Arduino framework apparently `define`s its own macro for `round()` - a simple `#undef round` should be able to fix that.

OttoWinter, 2018-01-25:
I believe I fixed the bug on the new `fix-cmath` branch. Could you, if you have time, download [this zip file](http://github.com/OttoWinter/esphomelib/zipball/fix-cmath/), extract it and run `platformio run` while inside the folder? Thanks. (Otherwise I can try to borrow a Windows laptop in a few days or so.)

Big4SMK, 2018-01-25:
Thank you for the detailed explanation. I have tried your code, and it works as expected. Great work!

OttoWinter, 2018-01-25:
Thank you for reporting this and helping with the testing! I hope my explanation can be understood -
 I'm known to sometimes express myself in a complicated manner 😂 . Closing this as it has been resolved.

closed 2018-01-25 by OttoWinter

Big4SMK, 2018-01-25:
I do have a little background in programming, that helps. Tomorrow I will see if I can get some simple code working. My quick test gives me some loglines before running into the following assertion, but first some sleep...
`assertion "(this)->get_component_state() == esphomelib::Component::SETUP || (this)->get_component_state() == esphomelib::Component::LOOP" failed: file ".piolibdeps\esphomelib_ID2096\src\esphomelib\component.cpp", line 61, function: void esphomelib::Component::loop_()
abort() was called at PC 0x400d131b on core 1
`

OttoWinter, 2018-01-25:
Yes, I should probably catch some sleep too. The assertion is probably because you forgot to call `app.setup()` at the end of your `setup()` function.

Big4SMK, 2018-01-26:
Spot on!