esphome/esphome-core #10: inverted led component

Schibum, 2018-03-04:
Hi,

not sure if I understand the 'inverted' property for the led output component correctly, but I think this:

https://github.com/OttoWinter/esphomelib/blob/95ff3365bbedf28cbe9be60201f16c3d383b700e/src/esphomelib/output/ledc_output_component.cpp#L21

should probably be 
`duty_non_inverted = max_duty - duty; `

Also the in pin 34 in the example:
https://github.com/OttoWinter/esphomelib/blob/95ff3365bbedf28cbe9be60201f16c3d383b700e/examples/livingroom.cpp#L21

might be confusing as GPIO pins 34-39 are input only on esp32.

OttoWinter, 2018-03-06:
> not sure if I understand the 'inverted' property for the led output component correctly, but I think this:

The reason I'm doing that there is for the [`ATXComponent`](https://github.com/OttoWinter/esphomelib/blob/master/src/esphomelib/atx_component.h), which can automatically enable a signal once a device requires high power output. The inverted option is only there to allow for inverted electronics, for example a p-channel fet. So if we set `inverted = True`, we'd like to output `1 - duty` to ledc, but only enable the ATX power supply if `duty > 0`. Looking at the code, I'm not entirely sure whether that's what I'm actually doing there... I will have a look at that on the week-end (for the last few weeks I've sadly not had too much time to work on my own projects, but I hope I can get back to improving esphomelib soon)

> Also the in pin 34 in the example:

Sure 👍, I'm working on the documentation for this library. There I'll also create a small all indicating all supported modes for each pin, since especially the ESP32 has some very weird behavior with some pins.

Schibum, 2018-03-18:
thx, what I actually meant was that this line should probably be `duty = max_duty - duty;` (updated original post), as `duty` is the value written to the pwm, so that's what needs to be inverted.

OttoWinter, 2018-03-23:
Thanks for reporting this! You were right, this should now be fixed with https://github.com/OttoWinter/esphomelib/pull/15

closed 2018-03-23 by OttoWinter