esphome/esphome-core #12: ATXComponent output level

Kyklas, 2018-03-11:
My hardware has power regulator that needs to be control as part of the light. The ATXComponent is exactly what I need.

Unfortunately, my regulator control is active high and the ATXComponent is active low.

Could the ATXComponent be based on gpio_binary_output_component so it can leverage the inverted feature ?

I can make the changes and do a pull request once design is approved.

OttoWinter, 2018-03-11:
> Could the ATXComponent be based on gpio_binary_output_component so it can leverage the inverted feature ?

I looked into it that some time ago, but I think there was a big issue with dependencies, so I just ignored the issue. Since then library has changed a lot though (about 2 rewrites in total 😅) so it might work now.

Kyklas, 2018-03-14:
I looked at the code and I just added a active_low property which is used when writing the value to the pin.

OttoWinter, 2018-03-23:
This now works with https://github.com/OttoWinter/esphomelib/pull/15 (and the power supply is active high by default now)

So now you should just be able to write

```cpp
App.make_power_supply(GPIOOutputPin(13, OUTPUT, true)); // active low
App.make_power_supply(GPIOOutputPin(13, OUTPUT, false)); // active high
App.make_power_supply(13); // active high, shorthand for above line
```

(this is not in 1.0.0, I think I'll release 1.1.0 with this patch today or tomorrow)

closed 2018-03-23 by OttoWinter