esphome/esphome-core #13: Question Deep Sleep integration ?

oncleben31, 2018-03-16:
Does this lib integrate or plan to implement deep sleep features to optimize energy consumption ?
Or perhaps we can manage it aside by adding the deep sleep code in the block below ?
```c
void loop() {
    app.loop();
}
```

OttoWinter, 2018-03-18:
When I designed the library (or rather framework) to be as easy as possible I intentionally made the decision to not support deep sleep. Don't get me wrong, it's great for battery-powered devices, but it sadly makes all other use cases much harder to handle. For example we would then need a way to somehow tell components that they should do all their work and report their values right away.

I mean it's on my task list, but it's less of a priority. I guess one (hackish) way to use deep sleep would be like this:

```cpp
#include <Esp.h>

# ....

int i = 0;
void loop() {
  if (i > 100) ESP.deepSleep(15*60*1000*1000);  // 15 minutes
  i++;
  app.loop();
  delay(10);
}
```

Or something like it; what it basically does is keep the device on for 100 `loop` cycles (should be enough to do all work, probably 10 cycles would also work) and then proceeds to sleep for some time.

oncleben31, 2018-03-19:
Okay, thanks for the explainations.

OttoWinter, 2018-04-18:
Deep Sleep Integration has arrived in version 1.3.0. Yay! https://esphomelib.com/esphomeyaml/components/deep_sleep.html

closed 2018-04-18 by OttoWinter

oncleben31, 2018-05-24:
Didn't take time to say thank you for implementing my feature request...
Thank you ;-)
Now it's done and I'm going to test the esphomeyaml addon.