esphome/esphome-core #14: Beginners question

ameerkamp, 2018-03-18:
I just installed Platformio and use your RGB example. When building there is a warning about ESPmDNS - do I have to fix that? 
I can also load the code to my board but Mosquitto Broker doen't show anything. Do I have to specify the host including port like this >app.init_mqtt("hassio.local:1883"<?
In the configuration I added >discovery: True< and >discovery_prefix: discovery<, is anything else needed?

OttoWinter, 2018-03-18:
> there is a warning about ESPmDNS - do I have to fix that?

Nope, that's just because of the library loading system and arduino-esp32.

> Do I have to specify the host including port like this >app.init_mqtt("hassio.local:1883")?

No, there's a second argument for [`init_mqtt`](https://github.com/OttoWinter/esphomelib/blob/95ff3365bbedf28cbe9be60201f16c3d383b700e/src/esphomelib/application.h#L81-L92) that you can use for ports. The call would then look like this:

```cpp
app.init_mqtt("hassio.local", 1883, "USERNAME", "PASSWORD");
```

(but port 1883 is the default anyway, so you can just leave the `1883` out)

> In the configuration I added >discovery: True< and >discovery_prefix: discovery<, is anything else needed?

No, that should do the trick.

> I can also load the code to my board but Mosquitto Broker doesn't show anything.

Hmm, that's weird... could you look at the serial monitor logs (with baud rate 115200) and see what's being outputted there? Thanks!

ameerkamp, 2018-03-19:
Thanks for the info. I loaded another software before from German CT Magazin. Maybe it's still in memory. I will try a total reset but that has to wait until the weekend.
-- 
Sent from my Android phone with WEB.DE Mail. Please excuse my brevity.

ameerkamp, 2018-03-24:
I now erased the board and loaded your example. I now get this output in the serial monitor:

I (3544) wifi: connected with FRITZBox_AME, channel 11
␛[0;37m[V][wifi_component.cpp:87] on_wifi_event(): [WiFi event] STA IsConnected (4)␛[0m
␛[0;37m[V][wifi_component.cpp:87] on_wifi_event(): [WiFi event] STA got IP (7)␛[0m
␛[0;37m[V][wifi_component.cpp:113] wait_for_connection(): .␛[0m
␛[0;37m[V][wifi_component.cpp:116] wait_for_connection():     WiFi connected.␛[0m
␛[0;37m[V][wifi_component.cpp:117] wait_for_connection():     IP Address: 192.168.178.63␛[0m
␛[0;36m[D][mqtt_client_component.cpp:29] setup(): Setting up MQTT...␛[0m
␛[0;37m[V][mqtt_client_component.cpp:33] setup():     Server Address: hassio.local:1883␛[0m
␛[0;37m[V][mqtt_client_component.cpp:34] setup():     Username: 'mqtt_admin'␛[0m

␛[0;37m[V][mqtt_client_component.cpp:35] setup():     Password: 'Ha-waii1'␛[0m
␛[0;37m[V][mqtt_client_component.cpp:37] setup():     Client ID: 'livingroom-30aea422a46c'␛[0m
␛[0;37m[V][mqtt_client_component.cpp:103] reconnect(): Reconnecting to MQTT...␛[0m
␛[0;37m[V][mqtt_client_component.cpp:106] reconnect():     Attempting MQTT connection...␛[0m
I (6518) wifi: pm start, type:0

␛[0;33m[W][mqtt_client_component.cpp:130] reconnect():     failed, rc=-2␛[0m
␛[0;33m[W][mqtt_client_component.cpp:131] reconnect():     Try again in 1 second␛[0m
␛[0;37m[V][mqtt_client_component.cpp:106] reconnect():     Attempting MQTT connection...␛[0m

So in Homeassistant I can't see it while my Sonoff Switch with tasmota software works fine.

OttoWinter, 2018-03-24:
Hmmm, it seems like the MQTT broker isn't accepting your connection. At least PubSubClient (the library that's used here to send/receive MQTT messages) is saying there's a return code `-2` (`MQTT_CONNECT_FAILED`). It's especially puzzling to me since Sonoff-Tasmota and esphomelib both use PubSubClient under the hood...

From what I see, you're using `hassio.local` as the server address. Could you maybe check for me if there's something in the hass.io logs. Could you maybe also try using an IP instead of `hassio.local`?

Also, I see your serial monitor isn't set up to handle the color codes (`[0;37m` and so on) . If you're on a unix machine, you can use `platformio device monitor --raw [...]` to have beautiful color coded log messages like this:
[image]

ameerkamp, 2018-03-24:
After using the ip adress it works somehow:

1521890939: New connection from 192.168.178.63 on port 1883.
1521890939: New client connected from 192.168.178.63 as livingroom-30aea422a46c (c1, k15, u'mqtt_admin').
1521891547: New connection from 192.168.178.63 on port 1883.
1521891547: Client livingroom-30aea422a46c already connected, closing old connection.
1521891547: Client livingroom-30aea422a46c disconnected.
1521891547: New client connected from 192.168.178.63 as livingroom-30aea422a46c (c1, k15, u'mqtt_admin').

But I can't see it in Homeassistant. My Sonoff is shown in the Overview as well as in the States window.

OttoWinter, 2018-03-24:
Ok, some progress 😅 (I will look into the IP thing later, it should definitely be fixed)

First, what version of `esphomelib` are you using? In 1.1.0 I changed the default `discovery_prefix` to Home Assistant's default: `homeassistant/`, so you might need to remove `discovery_prefix` if you have updated the library with `platformio lib update`.

Next, do you have a program with which you can see the MQTT messages under a specific topic? Then you could see what's under `homeassistant/+/livingroom/#` (or `discovery/+/livingroom/#` if you're using 1.0.0)

Thanks very much!

ameerkamp, 2018-03-25:
I think its something within my Hassio installation. In the log I find these messages and I will try to fix these first before another attempt with your lib:

2018-03-25 15:20:03 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: (sqlite3.DatabaseError) database disk image is malformed (Background on this error at: http://sqlalche.me/e/4xp6) (retrying in 3 seconds)
2018-03-25 15:20:06 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: (sqlite3.DatabaseError) database disk image is malformed (Background on this error at: http://sqlalche.me/e/4xp6) (retrying in 3 seconds)
2018-03-25 15:20:09 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: (sqlite3.DatabaseError) database disk image is malformed (Background on this error at: http://sqlalche.me/e/4xp6) (retrying in 3 seconds)
2018-03-25 15:20:11 WARNING (MainThread) [homeassistant.setup] Setup of recorder is taking over 10 seconds.
2018-03-25 15:20:12 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: (sqlite3.DatabaseError) database disk image is malformed (Background on this error at: http://sqlalche.me/e/4xp6) (retrying in 3 seconds)
2018-03-25 15:20:15 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: (sqlite3.DatabaseError) database disk image is malformed (Background on this error at: http://sqlalche.me/e/4xp6) (retrying in 3 seconds)
2018-03-25 15:20:19 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: (sqlite3.DatabaseError) database disk image is malformed (Background on this error at: http://sqlalche.me/e/4xp6) (retrying in 3 seconds)
2018-03-25 15:20:22 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: (sqlite3.DatabaseError) database disk image is malformed (Background on this error at: http://sqlalche.me/e/4xp6) (retrying in 3 seconds)
2018-03-25 15:20:25 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: (sqlite3.DatabaseError) database disk image is malformed (Background on this error at: http://sqlalche.me/e/4xp6) (retrying in 3 seconds)
2018-03-25 15:20:28 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: (sqlite3.DatabaseError) database disk image is malformed (Background on this error at: http://sqlalche.me/e/4xp6) (retrying in 3 seconds)
2018-03-25 15:20:31 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: (sqlite3.DatabaseError) database disk image is malformed (Background on this error at: http://sqlalche.me/e/4xp6) (retrying in 3 seconds)
2018-03-25 15:20:31 ERROR (MainThread) [homeassistant.setup] Setup failed for recorder: Component failed to initialize.
2018-03-25 15:20:31 ERROR (MainThread) [homeassistant.setup] Unable to setup dependencies of history. Setup failed for dependencies: recorder
2018-03-25 15:20:32 ERROR (MainThread) [homeassistant.setup] Setup failed for history: Could not setup all dependencies.
2018-03-25 15:20:33 ERROR (MainThread) [homeassistant.setup] Unable to setup dependencies of logbook. Setup failed for dependencies: recorder
2018-03-25 15:20:33 ERROR (MainThread) [homeassistant.setup] Setup failed for logbook: Could not setup all dependencies.
2018-03-25 15:20:53 WARNING (SyncWorker_12) [netdisco.ssdp] Found malformed XML at http://192.168.178.32:9080: status=ok

ameerkamp, 2018-03-25:
I fixed that but it doesn't change anything. In the Mosquitto log I see that livingroom-30aea422a46c is found but in the states windows it doesn't show up. I will give up on this for now and maybe reinstall hassio at a later point in time. Thanks for your help!

PR #30 (2018-03-27): Fix PubSubClient issue
See https://github.com/knolleary/pubsubclient/issues/375; PubSubClient doesn't copy the domain name and therefore messes up our connection code.

Took a while to debug, but now with a proper debugger it's just so much easier 👍

Fixes #14
closed 2018-03-27 by OttoWinter