esphome/esphome-core #25: Colour temperature light support?

messismore, 2018-03-24:
Does the library support colour temperature (= dual channel) lights? Or are there plans to do so?

OttoWinter, 2018-03-24:
Yes, I was planning on eventually adding those, but I don't own any hardware for those so I kind of kept away from adding support.

I think it should be relatively easy to do so I could add support for it rather quickly. I'm just wondering how the interface for these color temperature lights usually look like. Do they expose two channels (one for cold white, one for warm white) that you can send PWM signals to? Do these exist together with RGB too, so RGBWW?

messismore, 2018-03-24:
From what I understand dimmers expose two channels. I don't own any hardware yet either, but I haven't found dimmers yet that support a pwm input signal. I was hoping to be able to use something like [this](https://shop.deko-light.com/detail/index/sArticle/36760) or [that](https://www.barthelme.de/content/de/led/family/1624/chromoflex-pro-dmx-cv.aspx). 
[image]

Unfortunately it appears that the only way to control both channels is via DMX and I don't know how much of a headache that is.

messismore, 2018-03-25:
Okay it seems that Home Assistant indeed supports setting a colour temperature for RGBWW lights. [(See LimitlessLED)](https://www.home-assistant.io/components/light.limitlessled/)
Home Assistant [expects](https://www.home-assistant.io/components/light/) colour temperatures to be set in [Mireds](https://en.wikipedia.org/wiki/Mired), so I guess esphomelib would need to know the colour of each channel and mix accordingly?

OttoWinter, 2018-03-25:
So I did some more digging into this DMX format yesterday and fortunately it's supported by [FastLED](https://github.com/FastLED/FastLED/wiki/Chipset-reference#other-stuff) which I want to add suport for soon (quite a few people seem to be using addressable lights). Then that controller would hopefully also be supported.

About the RGBWW lights: Yes it's true we would need to do some color mixing ourselves, but that probably wouldn't be that big of a problem - looking at the Home Assistant source LimitlessLEDs also don't do nothing more than a bit of mixing.

messismore, 2018-03-25:
> About the RGBWW lights: Yes it's true we would need to do some color mixing ourselves, but that probably wouldn't be that big of a problem - looking at the Home Assistant source LimitlessLEDs also don't do nothing more than a bit of mixing.

Yeah it's just something to keep in mind since these tuneable white led strips all support different colour ranges. So it's not as simple as to just 'pan' between the two channels, one would also need to know which colour temperature each ratio belongs to.

OttoWinter, 2018-05-16:
So I've now implemented support for FastLED, however, support for DMX is going to be quite difficult.

The Arduino library FastLED uses for DMX only works for AVR (so no espressif) devices. I'll maybe try to port that library to esphomelib one day, but tbh it's not that big of a priority for me right now.

messismore, 2018-05-19:
That is unfortunate, but understandable. This might be the time for me to finally dabble in C…

Skaronator, 2018-07-02:
Would be cool to see support for RGBWW (aka. RGB+CCT) 5 channel light.

OttoWinter, 2018-09-27:
Done in 1.8.0 :)

closed 2018-09-27 by OttoWinter