esphome/esphome-core #26: Compile error ESP8266

geelenbert, 2018-03-25:
Hi,

I'm trying to build something for an NodeMCU v2 device. 

```
#include "esphomelib/application.h"

using namespace esphomelib;

void setup() {
  App.set_name("displates");
  App.init_log();

  App.init_wifi("xxxxxx", "xxxxxxxx");
  App.init_mqtt("192.168.1.xxx", "homeassistant", "xxxxxxxxxxx");
  auto *ota = App.init_ota();
  // Set a plaintext password, alternatively use an MD5 hash for maximum security (set_auth_password_hash)
  ota->set_auth_plaintext_password("knippen5");
  ota->start_safe_mode();

  App.make_monochromatic_light("Displate 1", App.make_ledc_output(13)); // supports brightness
  
  App.setup();
}

void loop() {
  App.loop();
}
```

Compiling the code gives:
```
[Sun Mar 25 19:39:21 2018] Processing displates (platform: espressif8266; board: nodemcuv2; framework: arduin
o)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf

LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> v2.6
|-- <ArduinoJson> v5.13.1
|-- <DHT> v1.0.0
|-- <DallasTemperature> v3.8.0
|   |-- <OneWire> v2.3.4
|-- <PCA9685 16-Channel PWM Driver Module Library> v1.2.9 #22b7212
|   |-- <Wire> v1.0
|-- <ESP8266WiFi> v1.0
|-- <ArduinoOTA> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
Compiling .pioenvs/displates/src/examples/displates.cpp.o
Compiling .pioenvs/displates/src/src/esphomelib/application.cpp.o
Compiling .pioenvs/displates/src/src/esphomelib/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/displates/src/src/esphomelib/binary_sensor/mqtt_binary_sensor_component.cpp.o
examples/displates.cpp: In function 'void setup()':
examples/displates.cpp:26:50: error: 'class esphomelib::Application' has no member named 'make_ledc_output'
App.make_monochromatic_light("Displate 1", App.make_ledc_output(13)); // supports brightness
^
Compiling .pioenvs/displates/src/src/esphomelib/component.cpp.o
*** [.pioenvs/displates/src/examples/displates.cpp.o] Error 1
========================== [ERROR] Took 4.61 seconds ==========================
Environment displates           [ERROR]
========================== [ERROR] Took 4.61 seconds ==========================
 
================================== [SUMMARY] ==================================
Environment livingroom          [SKIP]
Environment dht-dallas-sensors  [SKIP]
Environment switch-binarysensor [SKIP]
Environment fan-example         [SKIP]
Environment lights              [SKIP]
Environment livingroom8266      [SKIP]
Environment custombmp180        [SKIP]
```

If i replace the dimmable light with a binairy light, the code compiles without issues.
```
#include "esphomelib/application.h"

using namespace esphomelib;

void setup() {
  App.set_name("displates");
  App.init_log();

  App.init_wifi("xxxxxx", "xxxxxxxx");
  App.init_mqtt("192.168.1.xxx", "homeassistant", "xxxxxxxxxxx");
  auto *ota = App.init_ota();
  // Set a plaintext password, alternatively use an MD5 hash for maximum security (set_auth_password_hash)
  ota->set_auth_plaintext_password("knippen5");
  ota->start_safe_mode();

 App.make_binary_light("Displate 1"", App.make_gpio_output(13));
  
  App.setup();
}

void loop() {
  App.loop();
}
```

Platform config is:
```
[env:displates]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps = ${common.lib_deps}
build_flags = ${common.build_flags}
build_unflags = ${common.build_unflags}
src_filter = ${common.src_filter} +<examples/displates.cpp>
```

What could it be ?

OttoWinter, 2018-03-25:
You're trying to create an LEDC output component (`App.make_ledc_output`), but those are only available on the ESP32 (http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/ledc.html).

LEDC is basically a hardware PWM generator, or a peripheral that can generate high frequency modulated signals right from the chip. Sadly the ESP8266 doesn't have any of those. For the ESP8266 there's currently only the PCA9685 chip (and software PWM, but that's not really suitable for RGB lights as it flickers a lot).

geelenbert, 2018-03-26:
Well that would explain a lot ....

But i'm comming fomr Easy ESP, where I used Software PWM. The hardware is fixed using the ESP outputs so i can't change to a hardware pwm chip anymore.  

I will have a swing at the software pwm