esphome/esphome-core #31: Calling deprecated function in ESP8266WiFi

sytone, 2018-03-30:
Have not had a chance to dig into see the problem and wanted to flag it to see if you already knew. 

```
Compiling .pioenvs\nodemcuv2\lib641\esphomelib_ID2096\esphomelib\wifi_component.cpp.o
.piolibdeps\esphomelib_ID2096\src\esphomelib\wifi_component.cpp: In member function 'virtual void esphomelib::WiFiComponent::setup()':
.piolibdeps\esphomelib_ID2096\src\esphomelib\wifi_component.cpp:29:29: warning: 'void ESP8266WiFiGenericClass::onEvent(WiFiEventCb, WiFiEvent_t)' is deprecated (declared at XXXXXX\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:54) [-Wdeprecated-declarations]
WiFi.onEvent(on_wifi_event);
^
```

OttoWinter, 2018-03-30:
Yes I know about the deprecation warning :) But I don't think the method will be removed anytime soon since many examples on the Internet are using it.

Also the alternative of using separate `std::function` objects for each callback type is quite memory expensive and would need special treatment for the ESP32 platform.

I will try to address it soon though 👍

OttoWinter, 2018-04-01:
Should be fixed now

closed 2018-04-01 by OttoWinter