esphome/esphome-core #32: Can't connect to WiFi network

warfair1337, 2018-03-30:
Hi. Been looking at this project for a while, and today I started to test it. Experiencing some issues.
I'm using the current library build from github, not the one published in PIO. Decided to start easy with the DHT module. But it keeps crashing. Anything I'm doing wrong, or is there something wrong in the code?

Thanks.

Using PIO with VS Code. Tested on both Wemos D1 and NodeMCU.

During build I get this warnings:

```
In file included from C:\Users\WaRFaiR\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:34:0,
from C:\Users\WaRFaiR\.platformio\lib\i2cdevlib\src\I2Cdev.h:52,
from C:\Users\WaRFaiR\.platformio\lib\i2cdevlib\src\BMP085.h:36,
from C:\Users\WaRFaiR\.platformio\lib\i2cdevlib\src\BMP085.cpp:33:
C:\Users\WaRFaiR\.platformio\lib\i2cdevlib\src\BMP085.cpp: In member function 'int32_t BMP085::getPressure()':
C:\Users\WaRFaiR\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/math.h:57:34: warning: overflow in implicit constant conversion [-Woverflow]
#  define NAN (__builtin_nanf(""))
^
C:\Users\WaRFaiR\.platformio\lib\i2cdevlib\src\BMP085.cpp:253:23: note: in expansion of macro 'NAN'
if (up == 0) return NAN;

C:\Users\WaRFaiR\.platformio\lib\esphomelib\src\esphomelib\wifi_component.cpp: In member function 'virtual void esphomelib::WiFiComponent::setup()':
C:\Users\WaRFaiR\.platformio\lib\esphomelib\src\esphomelib\wifi_component.cpp:30:29: warning: 'void ESP8266WiFiGenericClass::onEvent(WiFiEventCb, WiFiEvent_t)' is deprecated (declared at C:\Users\WaRFaiR\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:54) [-Wdeprecated-declarations]
WiFi.onEvent(on_wifi_event);
```

My program is here (main.cpp)

```
`#include "esphomelib/application.h"

using namespace esphomelib;

void setup() {
    App.set_name("testbuild");
    App.init_log();

    App.init_wifi("my.ssid", "mykey");
    App.init_mqtt("10.x.x.x", "xxx", "xxx");
    App.init_ota()->start_safe_mode();

    // This is required to set up the i2c bus. 14 is SDA pin and 27 is SCL pin.
    // You only need to call this once.
    // App.init_i2c(4, 5);

    // 0x48 is the default address when address pin is pulled low.
    // App.make_htu21d_sensor("HTU21D Temperature", "HTU21D Humidity");

    App.make_dht_sensor(12, "Livingroom Temperature", "Livingroom Humidity");

    App.setup();
}

void loop() {
    App.loop();
    delay(10); // this keeps the ESP32/ESP8266 from wasting power.
}
```

When looking at the serial output, it keeps crashing:

```
␛[0;32m[I][log_component:pre_setup:60]: Log initialized␛[0m
␛[0;35m[C][ota:start_safe_mode:142]: Safe mode enabled. There have been 0 suspected unsuccessful boot attempts.␛[0m
␛[0;32m[I][application:setup:25]: Application::setup()␛[0m
␛[0;35m[C][wifi:setup:29]: Setting up WiFi...␛[0m
␛[0;35m[C][wifi:setup:32]:     SSID: 'warfair.guest'␛[0m
␛[0;35m[C][wifi:setup:40]:     Hostname: 'testhtu21d'␛[0m
␛[1;31m[E][wifi:wait_for_connection:138]:     Can't connect to WiFi network, restarting...␛[0m

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
```

OttoWinter, 2018-03-30:
Great to see more people using this project 👍

First, you don't need to worry about the compile errors. They're just compiler warnings and don't really affect anything. Now to the not connecting part: Yes, I've also noticed it and have almost solved it.

The master github branch is not really meant to be used for setups and the latest commit has an issue that I've been trying to debug for the last ~9 hours now (debugging with the ESP8266 is a mess). Simply put, we're running out of RAM. I have a fix almost ready and will put it in the master branch in some time :)

For the time being, I would suggest you take the version from platformio like described in the [Installation Instructions](https://github.com/OttoWinter/esphomelib#setting-up-development-environment) (working on cleaning that up... but writing docs is so cumbersome...). Create a new folder and run `pio init` in the created folder, next put something like this in the `platformio.ini`

```ini
[env:test]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps = esphomelib
build_flags =
    -DMQTT_MAX_PACKET_SIZE=512
    -DARDUINOJSON_ENABLE_STD_STRING
```

warfair1337, 2018-03-30:
Thanks for your quick reply. I'll try using DHT with the PIO-version to verify. I only use bme/bmp280 and htu21d in my projects, that's why I wanted to use the current master version :-) But great project, I think for my kind of sensors this is way quicker than espeasy.

warfair1337, 2018-03-30:
Tried again after your latest updates, and things seems a bit better now. But something I observed along the way. I have issues connecting to Wifi when using an SSID where the PSK contains non-alphanumeric character. The first SSID has a PSK with the %-symbol, not allowing me to connect. I tried another network with a plain alphanumeric PSK, and it connected. It could not reach the mqtt-server, which is on another subnet. Tried creating a new SSID in the same subnet with the MQTT-server, and it connects to MQTT.

It seems there are routing/gateway issues in esphomelib, and of course non-alphanumeric PSK.

sytone, 2018-03-31:
I also found out the SSID is case sensitive :)

OttoWinter, 2018-04-01:
Hmmm, the issue about non-alphanumeric PSKs is weird; I'll look into it, but I think it would rather be a bug in the arduino core, not esphomelib... Thanks for reporting it 👍

About the subnets: I'm not a networking guy, but isn't the whole reason for subnets that they restrict access to devices within them 👀?

warfair1337, 2018-04-01:
I have experienced issues before with non-alpha characters in SSID, but in most projects I have tested this isn't an issue. 

Subnets are not necessarily there to restrict access, but they will be if there is no route between them. In my case I have a different wlan ssid for iot-devices placed in one subnet. I'm not that into the workings of arduino libraries yet. But dhcp offers a default gateway (router) which knows the routes to the subnets.

warfair1337, 2018-04-01:
Would it be possible to add some println that shows information received/applied for networking, ip, mask, dns, gateway etc?

OttoWinter, 2018-04-01:
Yes, that's exactly what I just did 5 minutes ago in my testing :) 😂

Another question: on what board was this? ESP8266/ESP32? I just tried SSID: `abc`, Password: `123%42abc` on my NodeMCU (ESP8266) and it seemed to work, I'll try with the ESP32 and some other PSKs next.

warfair1337, 2018-04-01:
Hmm, I must have had a typo or something earlier. It failed on two Wemos D1 and one NodeMCU. Just tested again on a Wemos and ESP32 and both works with %-character now. Sorry :flushed:

OttoWinter, 2018-04-01:
Great to hear 👍
I've now pushed a patch that improves WiFi&MQTT logging 😀

[image]

warfair1337, 2018-04-01:
Omg, no idea why everything suddenly works, but now i can also connect to mqtt from the ssid in the other subnet. Not sure what has been going on here, I can just assume a bad day where everything I touched went wrong :confused:
Again, sorry to waster your time with something that wasn't an issue.

Logging looks great 😄

OttoWinter, 2018-04-01:
No problem :) It's always nice seeing other people see this framework. And esphomelib probably still has a ton of bugs that I need to fix. But in the end, fixing all those bugs is also very important to move the project forward 👍

OttoWinter, 2018-04-18:
Closing this issue as the error is fixed from what I can tell from the comments

closed 2018-04-18 by OttoWinter

aracdin, 2018-06-18:
I'm currently running into problems connecting to wi-fi. Tried a build with an ESP-01 and a Huzzah and it never connects to my wi-fi. Build something in the Arduino IDE and it connects happily.

aracdin, 2018-06-18:
...so I've taken an Arduino IDE created program and built it in PlatformIO/Atom. Over the next few days, I'll try and get a minimum ESPHome program to build where I can "see" it and see what is different. One thing I have noticed - the Wifi setup features channel=-1 as a parameter, which I'm guessing just tells it to check all channels...

OttoWinter, 2018-06-22:
@aracdin Yes WiFi channel -1 means "use any channel". For the WiFi connectivity: Have you tried using the latest arduino framework version? Sometimes that helps (see `esphomeyaml:` section doc page and FAQ)

aracdin, 2018-06-23:
So, I tried to do a complete re-install of everything and having *one* platformio loaded under atom [because it's a nicer place to play :-)]. I symlinked platformio from pip freeze and pip list both show platformio==3.5.3 but pip install esphomeyaml doesn't find platformio, instead it complains
OSError: [Errno 13] Permission denied: '/usr/lib/python2.7/site-packages/platformio-3.5.3.dist-info'
which I guess is it wanting a quick sudo... So sudo it I hear the world cry. Then I get:
IOError: [Errno 40] Too many levels of symbolic links: '/usr/bin/platformio'

This feels less like python, and more like Ouroboros....still, I am learning a lot..... Question, (if you don't mind have you used esphomeyaml with platformio under Atom? Did it work?

brandond, 2018-06-25:
I think folks have (in general) stopped installing things system wide unless using the system package manager (rpm/apt/etc) and use virtualenv with pip instead. You might try that?