esphome/esphome-core #33: Panic MQTTSensorComponent

warfair1337, 2018-03-30:
Hi, another problem with htu21d and i2c. I'm wanting to use this project for a lot of htu21d sensors, and I tried getting this to work. But it keeps looping with a panic/crash.

Thanks.

Using PIO with VS Code. Tested on both Wemos D1 and NodeMCU.

During build I get this warnings:

```
Compiling .pioenvs\nodemcuv2\liba12\i2cdevlib\LM73.cpp.o
In file included from C:\Users\WaRFaiR\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:34:0,
from C:\Users\WaRFaiR\.platformio\lib\i2cdevlib\src\I2Cdev.h:52,
ffrom C:\Users\WaRFaiR\.platformio\lib\i2cdevlib\src\BMP085.h:36,
from C:\Users\WaRFaiR\.platformio\lib\i2cdevlib\src\BMP085.cpp:33:
C:\Users\WaRFaiR\.platformio\lib\i2cdevlib\src\BMP085.cpp: In member function 'int32_t BMP085::getPressure()':
C:\Users\WaRFaiR\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/math.h:57:34: warning: overflow in implicit constant conversion [-Woverflow]
#  define NAN (__builtin_nanf(""))
^
C:\Users\WaRFaiR\.platformio\lib\i2cdevlib\src\BMP085.cpp:253:23: note: in expansion of macro 'NAN'
if (up == 0) return NAN;

C:\Users\WaRFaiR\.platformio\lib\esphomelib\src\esphomelib\wifi_component.cpp: In member function 'virtual void esphomelib::WiFiComponent::setup()':
C:\Users\WaRFaiR\.platformio\lib\esphomelib\src\esphomelib\wifi_component.cpp:30:29: warning: 'void ESP8266WiFiGenericClass::onEvent(WiFiEventCb, WiFiEvent_t)' is deprecated (declared at C:\Users\WaRFaiR\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:54) [-Wdeprecated-declarations]
WiFi.onEvent(on_wifi_event);
```

My program is here (main.cpp)

```
`#include "esphomelib/application.h"

using namespace esphomelib;

void setup() {
    App.set_name("testbuild");
    App.init_log();

    App.init_wifi("my.ssid", "mykey");
    App.init_mqtt("10.x.x.x", "xxx", "xxx");
    App.init_ota()->start_safe_mode();

    // This is required to set up the i2c bus. 14 is SDA pin and 27 is SCL pin.
    // You only need to call this once.
    App.init_i2c(4, 5);

    // 0x48 is the default address when address pin is pulled low.
    App.make_htu21d_sensor("HTU21D Temperature", "HTU21D Humidity");

    // App.make_dht_sensor(12, "Livingroom Temperature", "Livingroom Humidity");

    App.setup();
}

void loop() {
    App.loop();
    delay(10); // this keeps the ESP32/ESP8266 from wasting power.
}
```

When looking at the serial output, it keeps crashing with a panic:

```
␛[0;32m[I][log_component:pre_setup:60]: Log initialized␛[0m
␛[0;35m[C][ota:start_safe_mode:142]: Safe mode enabled. There have been 0 suspected unsuccessful boot attempts.␛[0m

Panic C:\Users\WaRFaiR\.platformio\lib\esphomelib\src\esphomelib\sensor\mqtt_sensor_component.cpp:19 esphomelib::sensor::MQTTSensorComponent::MQTTSensorComponent(std::string, esphomelib::sensor::Sensor*)

ctx: cont
sp: 3fff4010 end: 3fff42c0 offset: 01a0

>>>stack>>>
3fff41b0:  3fff44d4 00000c51 00000c51 3fff4290
3fff41c0:  3fffaca4 3ffe89dc 4020adf0 4020afdc
3fff41d0:  3fffb54c 00000008 3fffb574 4020eca4
3fff41e0:  00000000 00000000 3fffb574 4020b54b
3fff41f0:  00000000 00000000 00000000 401004d8
3fff4200:  3fff428c 3fff25c4 3fffb574 4020340b
3fff4210:  3fff428c 3fffb508 3fffb524 40203242
3fff4220:  3fff428c 3fff25c4 3fffb524 402034cf
3fff4230:  3fffb524 3fff25c4 3fffb1ac 402017c8
3fff4240:  3fff4b00 3ffeb4ec 3fff4c0c 3fff3298
3fff4250:  00000000 3fff428c 3ffe89dc 4021c5c2
3fff4260:  3fff4294 3ffeb4ec 3fff25c4 3fff3298
3fff4270:  3fffdad0 3ffeb4ec 3fff25c4 40202827
3fff4280:  3fffb188 3fff4b00 feefeffe 3fffb0c0
3fff4290:  3fffb508 feefeffe feefeffe feefeffe
3fff42a0:  feefeffe 00000000 3fff3290 4020ee38
3fff42b0:  feefeffe feefeffe 3fff32a0 40100700
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
```

OttoWinter, 2018-03-30:
Thanks for reporting this too 👍

Oh... I forgot to initialize the HTU21D pointers [here](https://github.com/OttoWinter/esphomelib/blob/master/src/esphomelib/sensor/htu21d_component.cpp#L51) 🤕. (I don't have this sensor yet so I can't really test it; that's also why it's marked as experimental in the docs)

warfair1337, 2018-03-30:
Thanks again. I'll test htu21d for you :)
I have some esp32's around that I also can test.

warfair1337, 2018-03-31:
After successfully connecting wifi (using alphanumeric psk) I got some progress on the htu21d.
I tested with Adafruit HTU21D library to verify, and it works. 

I'm a little unsure what pins you are asking to define. Is it gpio-number (e.g 5 and 4 for NodeMCU) or pin-number (e.g D2 and D1 for NodeMCU). Must this be specified? What about defaulting to boards i2c-pins.

Here's serial log output. I can see the htu21d is connected, first temperature reading is always 128,9C. After that NaN on both temperature and humidity.

```
[C][sensor::htu21d:setup:55]: Setting up HTU21D...
[C][component:setup_:126]:     Update interval: 15000ms
[C][sensor::mqtt:setup:30]: Setting up MQTT Sensor 'HTU21D Temperature'
[C][sensor::mqtt:setup:32]:     Expire After: 450s
[C][sensor::mqtt:setup:33]:     Unit of Measurement: '°C'
[C][sensor::mqtt:setup:34]:     Accuracy Decimals: 0
[C][sensor::mqtt:setup:35]:     Icon: ''
[C][sensor::mqtt:setup:36]:     # Filters: 1
[C][sensor::mqtt:setup:30]: Setting up MQTT Sensor 'HTU21D Humidity'
[C][sensor::mqtt:setup:32]:     Expire After: 450s
[C][sensor::mqtt:setup:33]:     Unit of Measurement: '%'
[C][sensor::mqtt:setup:34]:     Accuracy Decimals: 0
[C][sensor::mqtt:setup:35]:     Icon: 'mdi:water-percent'
[C][sensor::mqtt:setup:36]:     # Filters: 1
[I][application:loop:50]: Running through first loop()
[I][application:loop:59]: First loop finished successfully!
[D][sensor::htu21d:update:68]: Got Temperature=128.9°C Humidity=nan%
[W][sensor::htu21d:update:78]: Invalid Humidity: nan%
[D][sensor::htu21d:update:68]: Got Temperature=nan°C Humidity=nan%
[W][sensor::htu21d:update:73]: Invalid Temperature: nan!C
[W][sensor::htu21d:update:78]: Invalid Humidity: nan%
```

warfair1337, 2018-03-31:
Also tested on the ESP32. I defined pins 21 SDA and 22 SCL which is the boards default.
Code is identical to what I used for the 8266, except sda/scl pins.
Failed the connection to htu21d. See output.

```
[C][ota:setup:22]: Setting up OTA...
[C][ota:setup:23]:     port: 3232
[C][ota:setup:28]:     hostname: 'testhtu21d'
[C][sensor::htu21d:setup:55]: Setting up HTU21D...
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 40
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 40
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 40
[E][sensor::htu21d:setup:60]: Connection to HTU21D failed.
[C][component:setup_:126]:     Update interval: 15000ms
[C][sensor::mqtt:setup:30]: Setting up MQTT Sensor 'HTU21D Temperature'
[C][sensor::mqtt:setup:32]:     Expire After: 450s
[C][sensor::mqtt:setup:33]:     Unit of Measurement: '°C'
[C][sensor::mqtt:setup:34]:     Accuracy Decimals: 0
[C][sensor::mqtt:setup:35]:     Icon: ''
[C][sensor::mqtt:setup:36]:     # Filters: 1
[C][sensor::mqtt:setup:30]: Setting up MQTT Sensor 'HTU21D Humidity'
[C][sensor::mqtt:setup:32]:     Expire After: 450s
[C][sensor::mqtt:setup:33]:     Unit of Measurement: '%'
[C][sensor::mqtt:setup:34]:     Accuracy Decimals: 0
[C][sensor::mqtt:setup:35]:     Icon: 'mdi:water-percent'
[C][sensor::mqtt:setup:36]:     # Filters: 1
[I][application:loop:50]: Running through first loop()
[I][application:loop:59]: First loop finished successfully!
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 40
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 40
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 40
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 40
[D][sensor::htu21d:update:68]: Got Temperature=nan°C Humidity=nan%
[W][sensor::htu21d:update:73]: Invalid Temperature: nan!C
[W][sensor::htu21d:update:78]: Invalid Humidity: nan%
```

OttoWinter, 2018-04-01:
>I'm a little unsure what pins you are asking to define. Is it gpio-number (e.g 5 and 4 for NodeMCU) or pin-number (e.g D2 and D1 for NodeMCU). Must this be specified? What about defaulting to boards i2c-pins.

esphomelib uses the GPIO pin numbers (i.e. 5 & 4) everywhere. Because of the magic of esphomelib (or to be more precise the arduino framework), you can type the following 🤓

```cpp
App.init_i2c(D2, D1);
# equivalent to: 
App.init_i2c(4, 5);
```

I'm not a huge user of the ESP8266 myself, but are D2 and D1 really always the default i2c pins on all boards? If not, I'd rather have users type out the full pin numbers instead of some users needing to debug the code because their board uses different default i2c pins...

Re HTU21D "Connection to HTU21D failed.": Hmmm, I don't know what could be going wrong here. I mean I'm just using an existing library ([i2cdevlib](https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/HTU21D)) for getting the sensor values here and I'd hope that that library works... 

Anyway, I think there might have been another issue in my own code (but I kind of doubt it). Otherwise I think I'll just re-write the library myself as I just did with the DS18b20 sensors (ended up making those sensors *much* more stable). The quality of the 3rd party libraries I've been using apparently seems to not be too good :/

warfair1337, 2018-04-01:
The htu21d connectes on esp8266, but fails to connect on the esp32. 

As for default pins, all pinout diagrams etc and papers i have read says that esp12-chip uses gpio 4 and 5 (d1 and d2 on nodemcu/wemos etc). for i2c, and 21 and 22 on the esp32-chip. This can be changed, but unless defined somewhere, it will use these default pins. That's my understanding, don't take my word for it 😆

OttoWinter, 2018-04-01:
Well, at least some progress 😺

That's really weird that it would only work on one board... I'll try my BMP180 (also i2c-based) on the ESP32 again too to maybe gain some more insight. And looking at the pin definitions, you're right that the SDA and SCL pins are well defined, I will put it in the code; thanks very much 👍

OttoWinter, 2018-04-18:
Ok, I think the issue should now be fixed with the sensor rewrite/refactor in 1.3.0. Closing this issue...

closed 2018-04-18 by OttoWinter