esphome/esphome-core #34: Lights show up as "Unavailable" in Homeassistant

geelenbert, 2018-03-30:
I've managed to get the ESP devices going for the most part i think. But the devices i create show up as "Unavailable" in home assitant. I just want to start with a couple of On Off leds.

my ESP code:

#include "esphomelib/application.h"

using namespace esphomelib;

```
void setup() {
  App.set_name("Displate Leds");
  App.init_log();

  App.init_wifi("xxxx", "xxxx");
  App.init_mqtt("xxx", "homeassistant", "xxxx");
  auto *ota = App.init_ota();
  // Set a plaintext password, alternatively use an MD5 hash for maximum security (set_auth_password_hash)
  ota->set_auth_plaintext_password("xxxxx");
  ota->start_safe_mode();

  App.make_binary_light("Displate 1", App.make_gpio_output(13));
  App.make_binary_light("Displate 2", App.make_gpio_output(5));
  App.make_binary_light("Displate 3", App.make_gpio_output(4));
  App.make_binary_light("Displate 4", App.make_gpio_output(15));
  App.make_binary_light("Displate 5", App.make_gpio_output(12));

  App.setup();
}

void loop() {
  App.loop();
}

```

Now the device and home assistant subscribe to the MQTT broker:
```
starting version 3.2.2
1522445452: mosquitto version 1.4.12 (build date 2017-06-01 13:03:46+0000) starting
1522445452: Config loaded from /etc/mosquitto.conf.
1522445452: Opening ipv4 listen socket on port 1883.
1522445452: Opening ipv6 listen socket on port 1883.
1522445452: Warning: Mosquitto should not be run as root/administrator.
1522445453: New connection from 172.30.32.1 on port 1883.
1522445453: New client connected from 172.30.32.1 as b0919dcd-0a78-4ad9-9aab-0c4c9de3f5ae (c1, k60, u'homeassistant').
1522445453: New connection from 192.168.1.67 on port 1883.
1522445453: New client connected from 192.168.1.67 as displates-5ccf7f024ecf (c1, k15, u'homeassistant').
1522445692: Socket error on client displates-5ccf7f024ecf, disconnecting.
1522445713: New connection from 192.168.1.67 on port 1883.
1522445713: New client connected from 192.168.1.67 as displate_leds-5ccf7f024ecf (c1, k15, u'homeassistant').
```

Home assistant finds the lights, but they appear as "Unavailable":
[image]

What could be the issue ? Can't find anything useful in the logs.

Also, how do i remove auto discovered items from Home assistant ?

OttoWinter, 2018-04-01:
The unavailable light thing is I think because of a bug that was in the `master` branch two days ago, but I think it should be fixed now ❤️ A good place to log for more information are always the MQTT topics `displate_leds/#` and `homeassistant/+/displate_leds/#` (the latter for Home Assistant MQTT discovery)

About the auto discovered items not disappearing: Yes, I know why that happens, but I sadly don't know how to fix it. See, to make Home Assistant "auto discover" an entity, esphomelib sends a MQTT message to the MQTT topic `homeassistant/sensor/heatpump/outside_temperature/config` with a payload like this:

```json
{
  "name": "Outside Temperature",
  "platform": "mqtt",
  "state_topic": "heatpump/sensor/outside_temperature/state",
  "availability_topic": "heatpump/status",
  "unit_of_measurement": "°C",
  "expire_after": 675
}
```

This tells Home Assistant that there's a sensor called "Outside Temperature" on the MQTT topic `heatpump/sensor/outside_temperature/state`. But in order for this discovery to work when Home Assistant is restarted, we have to *retain* those messages, i.e. tell the broker to re-send those messages to every client on each reconnect. Otherwise all your sensors wouldn't show up when Home Assistant restarts. But because these messages are retained, they will be re-sent even though the sensor (or light) doesn't exist anymore.

I have a small script that I will try to publish soon that will "clean up" all retained messages from MQTT discovery so that Home Assistant doesn't see them again, and I will try to publish it with the next esphomelib release (hopefully today or tomorrow)

OttoWinter, 2018-04-07:
Sorry for the delay, I've been caught up in real-life stuff a bit. Anyway... the clean script is now available in esphomeyaml: https://esphomelib.com/esphomeyaml/components/mqtt.html#using-with-home-assistant. See the `clean-mqtt` comment in there ❤️

OttoWinter, 2018-04-18:
So the script is now out here https://esphomelib.com/esphomeyaml/components/mqtt.html#using-with-home-assistant and it should help working around the issue. As there's no real way esphomelib can solve the problem I'm closing this issue

closed 2018-04-18 by OttoWinter