esphome/esphome-core #35: Custom sensor push new value error

ferazambuja, 2018-04-01:
First great work with the script and the documentation :)

I'm getting the following error when I try to compile:
_src/main.cpp:37:24: error: no matching function for call to 'HDC1080_Sensor::push_new_value(int&)'_
src/main.cpp:41:10: error: 'int8_t HDC1080_Sensor::accuracy_decimals()' marked 'override', but does not override
`
#include <Arduino.h>
#include <Wire.h>
#include "ClosedCube_HDC1080.h"
#include "esphomelib/application.h"

using namespace esphomelib;

class HDC1080_Sensor : public sensor::PollingSensorComponent {
 public:
  ClosedCube_HDC1080 hdc1080;

  HDC1080_Sensor(uint32_t update_interval) : sensor::PollingSensorComponent(update_interval) {}

  void setup() override {
    hdc1080.begin(0x40);
  }

  void update() override {
    int temp = (hdc1080.readTemperature());
    push_new_value(temp);
  }

  std::string unit_of_measurement() override { return "°C"; }
  int8_t accuracy_decimals() override { return 2; } // 2 decimal places of accuracy.
};

void setup()
{
  App.set_name("livingroom");
  App.init_log();

  App.init_wifi("YOUR_SSID", "YOUR_PASSWORD");
  App.init_ota()->start_safe_mode();
  App.init_mqtt("MQTT_HOST", "USERNAME", "PASSWORD");

  auto *custom_sensor = App.register_component(new HDC1080_Sensor(5000));
  App.make_mqtt_sensor_for(custom_sensor, "HDC1080_Sensor");

  App.setup();

}

void loop()
{
  App.loop();
}
`
plataformio.ini
`
[env:m5stack-core-esp32]
platform = espressif32
board = m5stack-core-esp32
framework = arduino
lib_deps =
  esphomelib
  ClosedCube_HDC1080
`

Same happens for the example:
error: no matching function for call to 'BMP180Sensor::push_new_value(double)'

Thanks

OttoWinter, 2018-04-01:
I just added support for HDC1080 sensors: https://github.com/OttoWinter/esphomelib/commit/779399fc075fdf01fc15598f120038a12e759674 🎉

If you can verify it works with your sensors, that would be great 👍. It's going to be in the v1.2 release. Hope you like it 😀

ferazambuja, 2018-04-01:
That was amazing!
It just works. 
Copied the code from the examples and removed the ones that I don't have :)

About the "essential" features I just got it and whatever is the default mode seems to work like a charm.

OttoWinter, 2018-04-04:
Closing this as v1.2 is out and the code should now be working.

closed 2018-04-04 by OttoWinter