esphome/esphome-core #36: Suggestion

WoodsterDK, 2018-04-03:
Hi there,

your project seems to be a great building block for adding devices to HA.

Have you thought about adding WifiManager capabilities - if no known network is found...

OttoWinter, 2018-04-04:
So what you're proposing is that if esphomelib can't establish a  connection, it would create  an wireless Access Point that one could connect to, right? Yes, I thought about that too, but honestly I don't really see a use case for it right now.

First up, currently esphomelib is written in a way that makes it depend on an active WiFi connection, and won't run any other component code as long as no connection to the MQTT broker is established. Code-Wise that makes dealing with hardware a lot easier. Let's look at some scenarios in which your proposed feature would be useful or not:

Scenario #.1: suppose we have a WiFi network called "abc" setup with esphomelib and for some reason the router is temporarily switched off. In this case, esphomelib will currently not do anything at all really. But as long as the network is switched off you wouldn't be able to do anything useful with the node anyway (as there is no connection to Home Assistant). So I think in this scenario the only reason I would see for an access point would be to see some logs from the node which would probably be very very difficult to do because of memory constraints.

Scenario #.2: Suppose the WiFi network "abc" that's setup in esphomelib has been renamed to "def". So as a user you'd want to upload a new firmware that has the new SSID. Unfortunately, because of the missing WiFi-Connection, OTA updates won't work. In this case a WiFi AP *would* be useful because then users would be able to fix  the SSID without having to use a USB cable. But because this scenario would only happen very seldomly I don't see why this would be that necessary.

Scenario #.3: Internal webserver. A feature I'm planning to add esphomelib at some point is an internal webserver that can be used for viewing entity states. This will however not include options to configure the node from within the browser - it's simply not what this framework is trying to do. But for the aforementioned state viewing front end, having an Access Point would be definitely useful.

On a technical level, a wireless access point would make many parts of the code a lot more difficult, as the code would need to be a lot more "real time" and this feature is not really a priority for me right now. Is there maybe another use case/scenario I missed?

messismore, 2018-04-04:
I see the benefit mainly for the initial setup (provide Wifi credentials and point to Home Assistant / MQTT broker) or when the Wifi changes for some reason (#.2).
This would make IOT devices built on esphomelib way more user friendly down the road. But I agree that this is a 'nice to have' feature that should not have a high priority right now.

OttoWinter, 2018-04-18:
Ok, while doing the web server I found out that an AP mode isn't all that difficult to do. Version 1.3.0 now includes an additional AP and AP+STA mode 🎉

https://esphomelib.com/esphomeyaml/components/wifi.html#access-point-mode

closed 2018-04-18 by OttoWinter

SirRedZ, 2019-01-23:
What is the IP address of the webserver? I tried 192.168.4.1 and 0.1 and 1.1 and no luck. I have checked the docs on Acess Point Mode and manually assigning an IP to ESP module is optional.

brandond, 2019-01-23:
@SirRedZ [192.168.4.1](https://github.com/OttoWinter/esphomelib/blob/dev/src/esphomelib/wifi_component.cpp#L877)

I believe this was carried over from the standard esp8266 arduino framework SoftAP configuration, but could probably be specifically called out in the esphomelib docs.

SirRedZ, 2019-01-24:
I did not enable the webserver component; as I looked it up and it didn't mention anything about allowing wifi credentials to be entered, Maybe I am asking for the wrong thing-- I want my freshly flashed module to present the AP (it currently does as required) and allow a simple interface to configure 2 or 3 SSID / Password to sequentially try upon save and reboot.

brandond, 2019-01-24:
It does not do that. There is no dynamic wifi configuration; if you want to change anything (including the wifi config) you have to recompile and re-flash. You can use the SoftAP network to upload your new configuration via WiFi OTA flash, though.

SirRedZ, 2019-01-24:
@brandond --So just to make sure I understand, you MUST input wifi credentials during 1st serial flash, otherwise there is no other way to connect a "fresh" node to your network after the fact - For instance preparing a node for a friend's home; not on your network = serial flash required (again).?!

Feature Request?> -- I loved being able to flash my tasmota units, and configure them to use WifiManager for connection info during 1st time setup. OTA afterwards in ESPHome is awesome !!!

Thanks Brandon!

brandond, 2019-01-24:
You can always configure multiple wifi networks; it will try them in the listed order. I'm not sure if @OttoWinter has dynamic wifi config on his roadmap, but at the moment everything is fixed in code.

OttoWinter, 2019-01-25:
@SirRedZ There are many reasons for doing it this way (see other issues in this repo for more info). So no, unless I find a better way there won't be such a feature - and also it would be against ESPHome's architecture: WYSIWYG.

And yes you can configure multiple wifi networks, and even have the ESP create an AP on its own as a fallback.

SirRedZ, 2019-01-25:
@OttoWinter  - thanks for taking time to reply. I think what you and your team have done is fantastic- I don't know coding or Arduino- so this gives me some confidence to attempt new things. (also thank you brandond).

alnino2005, 2019-05-22:
Slightly lost: What exactly is the AP mode used for? I saw mention that it can be used to OTA flash if the node is not connected to an AP...so how would that be done?

Im working on some Sonoff switches, and Im nervous about infrastructure changes while they're plastered inside of the wall. I'd like to be able to reliably OTA flash the units without having to ever touch them.

HFsi, 2019-05-22:
I'm in love with ESPhome, don't misunderstand me...but I have to say that it would be great to have the dynamic WiFi feature, as it is very useful for network changes