esphome/esphome-core #37: Publish processed values instead of raw sensor data

jjlawren, 2018-04-05:
I'd like to take raw sensor data, process it, and then send along derived values over MQTT. I don't want to publish raw distance sensor data on a periodic basis, but would rather take the data and only publish when the values cross some pre-defined threshold(s). For example, using a ceiling-mounted distance sensor to detect both if a garage door is opened/closed and if a vehicle is present.

OttoWinter, 2018-04-06:
That's exactly what [sensor filters](#_CPPv2N10esphomelib6sensor19MQTTSensorComponent17add_lambda_filterE15lambda_filter_t) are for in esphomelib 💪. They pre-process sensor values in order to decrease network traffic and put less pressure on Home Assistant's state machine.

I don't know the exact thresholds for your setup, but something like this might work:

```cpp
auto o = App.make_ultrasonic_sensor(...);
o.mqtt->clear_filters();
o.mqtt->add_lambda_filter([&](float value) -> Optional<float> {
  if (value > 2.0f || isnan(value))
    // Returning an empty optional stops the value from ebing published.
    return Optional<float>();
  ‎return value;
});
```

(writing this on my phone, as my laptop doesn't have any power; might not compile as-is)

Sensor filters are only supposed to do some basic pre-processing though, and more sophisticated stuff should be done with template (binary) sensors in Home Assistant. Home Assistant's template engine is amazing and it would be very difficult to create a similar, complete system for esphomelib.

OttoWinter, 2018-04-07:
And with esphomeyaml this has just gotten a lot easier: https://esphomelib.com/esphomeyaml/components/sensor/index.html#sensor-filters 😀