esphome/esphome-core #38: Light sensor support

ladefoged81, 2018-04-07:
Did I miss something or are no light sensors supported yet?
This would be a nice feature when using automatic lights.

ladefoged81, 2018-04-09:
I don't really have any suggestions as I don't really have any experience with any.
But I found this article and maybe some of them are easier to integrate than others?
https://www.intorobotics.com/common-budgeted-arduino-light-sensors/

OttoWinter, 2018-04-09:
Thanks for finding that link! Most of the sensors are already kind of supported through the ADC sensor and for the ESP32 the pulse counter sensor, but I will take a look at the i2c sensors. They probably won't be too difficult to implement 👍

ladefoged81, 2018-04-14:
When using a photoresistor or TEMT6000 (i'm currently testing with the last one) on ADC do you convert the voltage to lux when using it in HA?

OttoWinter, 2018-04-14:
In general, you'll have to consult the datasheet of the sensor. Every sensor has their own characteristic collector current to illuminance curve. Then you can use the [Lambda Sensor Filter](https://esphomelib.com/esphomeyaml/components/sensor/index.html#sensor-filters) to convert the incoming values to another unit, see [Converting Celsius to Fahrenheit](https://esphomelib.com/esphomeyaml/components/sensor/index.html#example-converting-celsius-to-fahrenheit) for an example of this. Above links are for esphomeyaml, if you're just using esphomelib I can also show you have it would be done in there.

Typically you'll hook up a photoresistor as a voltage divider using a known value resistor. On the [TEMT6000 Sparkfun Board](https://learn.sparkfun.com/tutorials/temt6000-ambient-light-sensor-hookup-guide) this resistor has a value of 10kΩ. The [datasheet of the TEMT6000](https://www.sparkfun.com/datasheets/Sensors/Imaging/TEMT6000.pdf) has a figure that shows you the relationship between current and illuminance in lux. With the 10kΩ series resistor and `VCC` hooked up to 3.3V, the current across the TEMT6000 should be about `adcvalue * 3.3 / 10000.0`. And based on the datasheet there's a proportional relationship between current and illuminance: every 2 microamps in current correspond to 1 lux. Therefore the formula should be `(adcvalue * 3.3 / 10000.0) * 2000000.0`.

So if using esphomeyaml, the configuration would be something like this:

```yaml
sensor:
  - platform: adc
    pin: A0
    name: Illuminance
    unit_of_measurement: lux
    filters:
      - lambda: (x * 3.3 / 10000.0) * 2000000.0
```

----

I'm not 100% sure how esphomelib/esphomeyaml should handle these 2nd order sensors that use the ADC. It would be awesome if we could have predefined components like the TEMT6000 that use the existing components (here ADC) and setup all filters for the user. However that would make the project a lot bigger and less maintainable :/ So I guess one option would be to do all of this 2nd order component stuff within esphomeyaml. If you can verify that above filter value conversion works, that would be awesome 😀

ladefoged81, 2018-04-14:
My suggestion would be to recommend users to use a sensor that works out-of-the-box and reports an actual lux value to make it easy for them.
If one or two i2c sensors is supported by esphomelib and we know they work that would be great.
Maybe ADC should be for "advanced use/users only"?
I'm not very experienced with any of the sensors i'm actually using and most code is copy-paste from somewhere :-)

I've tried the formula for TEMT6000 and compared the results with an Aeon Labs MultiSensor 6.
Same room and same location (inside living room, cloudy day):

MS6: 15lux
EHL: 5lux

MS6: 11lux
EHL: 4lux

OttoWinter, 2018-04-22:
Thanks! It will require a bit of cleaning up but would be relatively easy to add support for. I'll have to take a look at this in a few days as I currently don't have too much time on my hand.

ladefoged81, 2018-04-22:
I've ordered some BH1750 and TSL2561 (both i2c) on Aliexpress a week ago but it takes forever to delivered.
I've currently only have one TEMT6000 analog sensor to play with :grinning:

ladefoged81, 2018-05-03:
I'm still only using the one TEMT6000 sensor I have but it works good enough so i've removed the Aeon Labs MultiSensor 6 from my setup inside the house.
Regarding the values calculated with TEMT6000:
I don't think they a very accurate with the ADC and seems to cap at 726 lux so a digital sensor would be nice.
I've set my indoor light to turn on below 20lux (combined with a PIR sensor) and that works nicely.

ladefoged81, 2018-05-08:
Here is an example where it caps at 726 lux.
For use with automatic light this is not an issue but nice to know.

[image]

ladefoged81, 2018-05-08:
That would be nice. I've recieved 5 TSL2561 sensors the other day.
The board with the TEMT6000 is a ESP32 so I'll try set the attenuation to 11db and see what happens.

Totally unrelated question:
Where is the best place for general discussion on esphomelib/esphomeyaml? The Home Assistant forum?

OttoWinter, 2018-05-11:
So to your last question: I don't really know myself. The problem is that for me keeping track of all the Github issues (even though there are only ~10) is quite a pain, even with e-mail notifications.

The positive side of using the hass community forums is that it's a lot more real time and I can keep track of all threads much easier.

I've now also done a big i2c update ([here](https://github.com/OttoWinter/esphomelib/tree/i2c-update)) that will be released with 1.5.0. It includes (experimental) support for the TSL2561, BH1750, BME280, BME680 and SHT3XD 🎉

ladefoged81, 2018-05-15:
No luck with the TSL2561. I haven't recieved the BH1750 sensors yet.
The TSL2561 sensor shows up on a i2c scan with address 0x39, but I keep getting the same value 2.8lx (actually 2.814720). I have only tested with one TSL2561 sensor for now.

From the log:
[20:41:21][I][i2c:051]: Scanning i2c bus for active devices...
[20:41:21][I][i2c:057]: Found i2c device at address 0x39
...
[20:41:22][D][sensor.tsl2561:129]: Got illuminance=2.8lx
[20:41:22][D][sensor.mqtt:039]: 'ESP test TSL2561': Pushing out value 2.814720 with accuracy 1
[20:41:35][D][sensor.tsl2561:129]: Got illuminance=2.8lx
(same values repeat)

ladefoged81, 2018-05-15:
I have no idea what I'm looking at :grinning: 
Broken sensor is also a possibility. I can test with another, but not today.
```
[21:07:36][VV][component:107]: Running interval 'update':0 with interval=15000 last_execution=194969 (now=209970)
[21:07:36][VV][i2c:076]: Beginning Transmission to 0x39:
[21:07:36][VV][i2c:109]:     Writing 0b00000000 (0x00)
[21:07:36][VV][i2c:109]:     Writing 0b00000011 (0x03)
[21:07:36][VV][i2c:081]:     Transmission ended. Status code: 0x00
[21:07:36][V][component:054]: set_timeout(name='illuminance', timeout=422)
[21:07:37][VV][component:107]: Running timeout 'illuminance':1 with interval=422 last_execution=209997 (now=210420)
[21:07:37][VV][i2c:076]: Beginning Transmission to 0x39:
[21:07:37][VV][i2c:109]:     Writing 0b00001100 (0x0C)
[21:07:37][VV][i2c:081]:     Transmission ended. Status code: 0x00
[21:07:37][VV][i2c:103]: Requesting 2 bytes from 0x39:
[21:07:37][VV][i2c:133]:     Received 0b00001100 (0x0C)
[21:07:37][VV][i2c:133]:     Received 0b00000010 (0x02)
[21:07:37][VV][i2c:076]: Beginning Transmission to 0x39:
[21:07:37][VV][i2c:109]:     Writing 0b00001110 (0x0E)
[21:07:37][VV][i2c:081]:     Transmission ended. Status code: 0x00
[21:07:37][VV][i2c:103]: Requesting 2 bytes from 0x39:
[21:07:37][VV][i2c:133]:     Received 0b00001110 (0x0E)
[21:07:37][VV][i2c:133]:     Received 0b00000010 (0x02)
[21:07:37][VV][i2c:076]: Beginning Transmission to 0x39:
[21:07:37][VV][i2c:109]:     Writing 0b00000000 (0x00)
[21:07:37][VV][i2c:109]:     Writing 0b00000000 (0x00)
[21:07:37][VV][i2c:081]:     Transmission ended. Status code: 0x00
[21:07:37][D][sensor.tsl2561:129]: Got illuminance=2.8lx
[21:07:37][V][sensor.sensor:022]: 'ESP test TSL2561': Received new value 2.814720
[21:07:37][V][sensor.sensor:028]: 'ESP test TSL2561':  Filter #0 aborted chain
```

OttoWinter, 2018-05-16:
Thanks for the logs! I've now identified at least one issue using them. Basically, for the implementation I was mostly following the datasheet which had an implementation detail only very hidden.

I'll try to fix the source code for the TSL2561 today and get it into 1.5.1 :)

ladefoged81, 2018-05-26:
I recieved the BH1750 sensors and it seem to work.  Getting some realistic values. :grin:

ladefoged81, 2018-05-26:
It's definitely better than using the TEMT6000 with the ADC in low light conditions.
The last 2 hours the TEMT6000 value was 0 in my living room and the BH1750 about 30 down to 7 (low light but with 2 lamps turned on).

OttoWinter, 2018-09-27:
Ok, all your suggested sensors are now supported and should work. Please open a new issue if one of them doesn't work.

closed 2018-09-27 by OttoWinter

ladefoged81, 2018-09-27:
I haven't tested TSL2561 recently, but BH1750 works like a charm.

2016for, 2019-03-17:
@ladefoged81 Hey I have been trying to get my TEMT6000 working but it seems like the Voltage readings from the sensor are pretty low (0.15V in a decently lit room) (which corresponds to 3lux or so)

What calculation did you use?

Did you also face the problem of exceptionally low values?

I just want to use it to see if it's somewhat dark to turn on light.

Thanks.

ladefoged81, 2019-03-17:
@2016for I used the code provided by Otto however I have moved to a bh1750 i2c sensor instead.