esphome/esphome-core #42: Compiler error wire.h not found

osmantuna, 2018-04-20:
Compiling .pioenvs\nodemcuv2\libc8a\I2Cdevlib-esphomelib_ID3838\AD7746.cpp.o
In file included from .piolibdeps\PCA9685 16-Channel PWM Driver Module Library\PCA9685.cpp:25:0:
.piolibdeps\PCA9685 16-Channel PWM Driver Module Library\PCA9685.h:61:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

#include <Wire.h>
^
compilation terminated.
*** [.pioenvs\nodemcuv2\lib000\PCA9685 16-Channel PWM Driver Module Library\PCA9685.cpp.o] Error 1
In file included from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\AD7746.h:38:0,
from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\AD7746.cpp:35:
.piolibdeps\I2Cdevlib-esphomelib_ID3838\src\I2Cdev.h:53:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

#include <Wire.h>
^
compilation terminated.
*** [.pioenvs\nodemcuv2\libc8a\I2Cdevlib-esphomelib_ID3838\AD7746.cpp.o] Error 1

OttoWinter, 2018-04-21:
Hmmm, `Wire.h` should be automatically included by the arduino framework. Unfortunately, this error is hard to diagnose without any information about the build system you're using. Could you provide some information about how you're compiling, what your `platformio.ini` looks like, what version of platformio/esphomelib you're using. Thanks

ladefoged81, 2018-04-22:
I've had some similar errors when compiling some of the code on my Windows computer.
I've think platformio fails to download some of the required files (from github?) because running the compile command again fixed it for me. It happened 2-3 times.

osmantuna, 2018-04-22:
I tried couple times and previous versions too. No luck for now. I will try on my personal pc and let you know

OttoWinter, 2018-05-16:
esphomeyaml now automatically includes all the internal arduino-espressif packages, so I think this error should now be fixed ... at least I hope so.

While debugging the issue, I noticed that sometimes this issue seems to also be fixed by running `platformio -d <name_of_node> -t clean` and deleting the `<name_of_node>/.pio*` cache folders.