esphome/esphome-core #43: Compiling DHT for ESP8266 fails, because of OTA stuff

simonszu, 2018-04-24:
Relevant YAML-configuration, (besides the wizard-autogen-stuff):
```
ota:
  password: 'changeme'

sensor:
  - platform: dht
    pin: D4
    temperature:
      name: "Serverschrank Temperatur"
    humidity:
      name: "Serverschrank Luftfeuchtigkeit"
    update_interval: 15s
```
Relevant output of the run command:
```
Compiling .pioenvs/serverschrankklima/lib959/esphomelib/esphomelib/sensor/ads1115_component.cpp.o
Compiling .pioenvs/serverschrankklima/lib959/esphomelib/esphomelib/sensor/bmp085_component.cpp.o
Compiling .pioenvs/serverschrankklima/lib959/esphomelib/esphomelib/sensor/dallas_component.cpp.o
.piolibdeps/esphomelib/src/esphomelib/ota_component.cpp: In member function 'virtual void esphomelib::OTAComponent::setup()':
.piolibdeps/esphomelib/src/esphomelib/ota_component.cpp:40:18: error: 'class ArduinoOTAClass' has no member named 'setPasswordHash'
ArduinoOTA.setPasswordHash(this->password_.c_str());
^
*** [.pioenvs/serverschrankklima/lib959/esphomelib/esphomelib/ota_component.cpp.o] Error 1
```

Removing the OTA part of the config does not fix this issue. 

Otherwise, great piece of work, i appreciate very much. Sometimes i'm just too lazy to code my own firmware ;)

OttoWinter, 2018-04-24:
I think this might be because you're using an old version of the Arduino espressif8266 framework (like mid-2016 old). [`setPasswordHash`](https://github.com/esp8266/Arduino/commit/e83f30a78d3cb6adb3db702706006e5036706a50) was only added in July of 2016 so that might be it. To fix the issue, please try running

```bash
platformio update
```

to install the latest (and greatest) version of platformio and the `arduino-espressif8266` framework. If it doesn't work, please feel free to ask again :)

I'll probably set up an FAQ soon that might help with these issues

simonszu, 2018-04-24:
Indeed. Looks like i had an old platformio version i have used for flashing a Tasmota some time ago. Updating fixed this.

closed 2018-04-24 by simonszu

untergeek, 2018-07-17:
This is still biting people (myself included). It might be nice to have a blurb about this somewhere in the documentation.  A FAQ or something.