esphome/esphome-core #44: Touch sensor support (esp32)

ladefoged81, 2018-04-25:
Maybe capacitive touch sensor input on the esp32 would be a nice feature to support?

Some info:
http://www.instructables.com/id/IOT-Made-Simple-Playing-With-the-ESP32-on-Arduino-/

OttoWinter, 2018-05-01:
Sorry for the late reply, somehow this issue/feature request didn't get to my notification center.

Anyway, yes that's a feature I've been working on some time ago. And it would definitely be a good thing to support.

From an API perspective, there are a few problems with the touch sensor support that have delayed my progress on this:

 * The only good way to get notified of a "touch event" is through interrupts (polling takes ~5ms per touch pad, not good). The problem is that we're only notified when a touch *begins* and we don't know when the user lifts the finger from the touch pad again.
 * Another question is how to integrate this into esphomelib. My current implementation just exposes it as a binary sensor that gets enabled for a short period of time when a touch is detected. Ideally, an "event" domain would be better suited for this, but Home Assistant currently has no such MQTT platform.
 * The values obtained by the touch pads are very different depending on the touch pad used. Therefore there needs to be a lot of customizability. [esp-idf](https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/touch_pad.html) does offer this customizability with tons of options like threshold, filters, meas/idle time, attenuation, high/low voltage reference, charge/discharge speed. I will need to evaluate which ones are necessary for good calibration and which ones confuse the user and aren't required for calibration.

(above text is mostly just a small write-up of the problems. It always helps me to write down what needs to be done to make good solutions)

ladefoged81, 2018-05-03:
I can't really contribute with any technical stuff but i would imagine "a touch-event" and "a long touch-event" of some kind. Maybe a double-press?
An example could be for turning light on/off and a long-touch for diming lights.
I've just noticed it when reading this article: http://www.instructables.com/id/IOT-Made-Simple-Playing-With-the-ESP32-on-Arduino-/
Personally I prefer mechanical buttons because i've like the felling compared to touch, but since the esp32 have capacitive touch sensors it could be a nice feature.