esphome/esphome-core #47: Second load fails

rlust, 2018-05-10:
I got the first lr.yaml to load but now when I try to upload I get the following error:
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('\x1b')

I am using a ESP32 WEMOSBAT board.

This looks very interesting!

Nice work!

rlust, 2018-05-11:
I also tried the same process with a ESP8266 first load works second try fails to upload. Must be missing  something in the process. using the run command.

OttoWinter, 2018-05-11:
Yeah I've also seen this issue come up some time ago, but then it suddenly disappeared again.

My guess back then was that it was due to some socket not being closed in a *software* reset (not hardware reset) or some registers not being reset. I've done some updates to the OTA component since 1.4.0 and maybe (hopefully) it will fix it. At least from the testing I've done now I've not seen it again. I think I'll release 1.5.0 sometime this weekend (with a big i2c update)

Landrash, 2018-05-18:
Had the same problem on before 1.5 but have flashed 4 different devices with 1.5 and no problems so far.

rlust, 2018-05-25:
This was fixed with 1.5 and all is well!

Thanks!

OttoWinter, 2018-05-27:
Great!

closed 2018-05-27 by OttoWinter