esphome/esphome-core #49: [Feature-request] Redundant WiFi infrastructure

pluskal, 2018-05-20:
Do you plan to implement support for redundant WiFi connections? 

It would improve fault tolerance if ESP is able to connect to different AP if some issue with current is detected. 

Issues that come to mind are: 
- AP is unavailable, 
- AP  does not provide IP address (DHCP is malfunctioning), 
- the default gateway is not assigned, 
- DNS server is not assigned, 
- DNS server is not connectable, 
- MQTT server (if used) is not connectable.

OttoWinter, 2018-05-21:
Yes, this is certainly of the things esphomelib currently doesn't handle very well. There are, however, two problems which stop esphomelib from being a bit better about handling this:

 * For some reason, the ESPs sometimes like to just disconnect from WiFi and never re-connect again until a software restart happens. This is why the node restarts after a while if no connection is detected.
 * What *should* the device do when it has no MQTT/WiFi connection? I mean it can't push out any sensor values and cannot respond to any commands anyway, so the question is what should it do then.

The WiFi redundancy is already kind of possible with the [WiFi AP+STA Feature](https://esphomelib.com/esphomeyaml/components/wifi.html). Basically, when the WiFi connection to an external AP is dropped, esphomelib will automatically create an AP after a while.

The first few scenarios could be "fixed" by a Multi Wifi architecture as described [here](https://github.com/OttoWinter/esphomeyaml/issues/15). It's going to be a bit difficult to implement, but is a future goal.

pluskal, 2018-05-21:
Good points, I would say that Multi Wifi would be sufficient. If it can be triggered by common issues in connectivity. 

From my point of view, liveness of IoT devices should be monitored by a 3rd component that raises alert when the dead component is detected. So as long as the esphome device keeps connecting and reconnecting to multiple wifi APs after the issue on a network is solved, the infrastructure ends in working state.

Evanlec, 2018-05-23:
I would also very much like to see the ability to connect to multiple (i.e. 1 of X) wifi networks, that would be very handy for portable devices.

tgdavies, 2018-05-26:
I would like this too. In my use case I develop attached to one network, then when the device is in-situ it uses a different one -- I could jumper a GPIO and choose the connection, but it would be better if I could just ask it to try both.

OttoWinter, 2019-02-02:
Gonna close this. Automations can be written on the device now and WiFi has multiple networks to connect to etc.

closed 2019-02-02 by OttoWinter