esphome/esphome-core #52: Build instructions

tgdavies, 2018-05-26:
I would like to make a few changes to my fork of esphomelib, but while I'm a fairly experienced embedded developer I have had trouble finding out how to build esphomelib and get platformio to use it -- some quick build instructions in the README would be nice.

OttoWinter, 2018-05-27:
Will do! It's actually quite simple: just navigate to the esphomelib root (which contains `src`, `examples`, etc.) and run the following command:

```bash
pio init --ide <THE_IDE>
```

Valid options for IDE are: atom, clion, codeblocks, eclipse, emacs, netbeans, qtcreator, sublimetext, vim, visualstudio, vscode. (I have only tested with clion, sublime text and vim; It may not work with some of the other ones without an additional setup)

tgdavies, 2018-05-27:
When I run `pio init --ide vim` in the directory I cloned esphomelib into I see:
```
Linking .pioenvs/livingroom/firmware.elf
Calculating size .pioenvs/livingroom/firmware.elf
text	   data	    bss	    dec	    hex	filename
537954	 150452	  30144	 718550	  af6d6	.pioenvs/livingroom/firmware.elf
====================================================================== [SUCCESS] Took 8.84 seconds ======================================================================

=============================================================================== [SUMMARY] ===============================================================================
Environment livingroom         	[SUCCESS]
Environment dht-dallas-sensors 	[SKIP]
Environment switch-binarysensor	[SKIP]
Environment fan-example        	[SKIP]
Environment lights             	[SKIP]
Environment livingroom8266     	[SKIP]
Environment custombmp180       	[SKIP]
Environment i2c-sensors        	[SKIP]
Environment pcf8574            	[SKIP]
Environment fastled            	[SKIP]
====================================================================== [SUCCESS] Took 8.85 seconds ======================================================================
```

I also have a basic platformio project with a `platformio.ini` of
```
[env:lolin32]
platform = espressif32
board = lolin32
framework = arduino
lib_deps = esphomelib
```
How do I get that to build with the modified version of esphomelib from my clone directory?

OttoWinter, 2018-05-27:
Ok, when you have `lib_deps = esphomelib` it pulls in the latest esphomelib version from the platformio registry.

If you want to use a local version of esphomelib, you need to tell platformio to use it. To do this, there are two ways (neither probably works with windows, unfortunately):

a)

**`platformio.ini`**:

```
[env:lolin32]
platform = espressif32
board = lolin32
framework = arduino
lib_deps = 
```

and within your lolin32 project, run:

```bash
mkdir lib
ln -s /full/path/to/esphomelib  lib/esphomelib
```

b) Use an absolute path for `lib_deps`:

```ini
[env:lolin32]
platform = espressif32
board = lolin32
framework = arduino
lib_deps = file:///full/path/to/esphomelib
```

The problem with b) is that the latter *copies* the esphomelib directory and doesn't re-compile any updated files without a full clean+rebuild. a) on the other hand will automatically detect file changes in your esphomelib directory and compile the changes automatically.

tgdavies, 2018-05-27:
OK, that makes sense -- but I don't understand where my esphomelib build is going. I sit somewhere under ~/.platformio? (I'm on OS X)

Also did `pio init --ide vim` actually build all of esphomelib?

OttoWinter, 2018-05-27:
The build will go under `<YOUR_PROJECT>/.pioenvs` (and some other files `<YOUR_PROJECT>/.piolibdeps`). `~/.platformio` contains the toolchains and packages needed to build & upload the project.

You're right `pio init --ide vim` does seem to do a full build, while the same command for my IDE `pio init --ide clion` does not. platformio's build system is very complicated and I myself don't really understand it.

Also, once you've set up the project, you need to run `pio run` within your project directory to build the project, and use `pio run -t upload` to upload.

OttoWinter, 2018-09-27:
Better instructions are now in the docs under the contributing section. Closing...

closed 2018-09-27 by OttoWinter