esphome/esphome-core #53: PlatformIO instructions

vertiginous, 2018-06-02:
This probably isn't an issue for people familiar with PlatformIO but it's never mentioned in the readme that you need to install esphomelib in PlatformIO.  It's spelled out clearly for the Arduino IDE.  It would be helpful information for newcomers to the tool like me. Thanks!

OttoWinter, 2018-06-02:
Well you don't really have to "install" esphomelib. You only need to add it into `lib_deps` inside `platformio.ini`. And I think the README does state that:

> Then create a new project for an ESP32-based board (for example, `nodemcu-32s`). Then open up the newly created `platformio.ini` file and insert
>
> ```init
> ; ...
> platform = espressif32
> board = nodemcu-32s
> framework = arduino
> lib_deps = esphomelib
> ```

vertiginous, 2018-06-02:
Oh, i see!  I'm using an esp8266 based board and I was unsure if the last two lines were needed.  I think the way the markdown syntax is being displayed is confusing.  It's swallowing the ellipsis after "board = nodemcuv2".  When I look at the raw markdown it makes more sense.

commit 887530a9 (2018-06-05): Improve README ESP8266 instructions. Fixes #53
closed 2018-06-05 by OttoWinter