esphome/esphome-core #55: Panic when Cover is changed

brandond, 2018-06-04:
I have a switch that is toggled by a Cover. The ESP32 panics when the Cover is used:
```
[10:24:54][D][cover.mqtt:048]: 'Garage Door Opener': Stopping cover...
[10:24:54][D][binary_sensor.mqtt:066]: 'Garage Door Switch': Sending state ON
[10:24:55][D][binary_sensor.mqtt:066]: 'Garage Door Switch': Sending state OFF
[10:24:55]Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
[10:24:55]. Exception was unhandled.
[10:24:55]Core 1 register dump:
[10:24:55]PC      : 0x4014c0d6  PS      : 0x00060130  A0      : 0x800d2e7e  A1      : 0x3ffd2450
[10:24:55]A2      : 0x3ffcdf64  A3      : 0x00000000  A4      : 0x00000050  A5      : 0x3ffd2430
[10:24:55]A6      : 0x08fffffd  A7      : 0x00000000  A8      : 0x8014c772  A9      : 0x3ffd2430
[10:24:55]A10     : 0xa5a5a5a5  A11     : 0x00000000  A12     : 0x00000005  A13     : 0x00000020
[10:24:55]A14     : 0x00000020  A15     : 0x3ffcdf04  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
[10:24:55]EXCVADDR: 0xa5a5a5a5  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
[10:24:55]
[10:24:55]Backtrace: 0x4014c0d6:0x3ffd2450 0x400d2e7b:0x3ffd2470 0x4014c0da:0x3ffd2490 0x400d2e8b:0x3ffd24b0 0x400d5041:0x3ffd24d0 0x400d5131:0x3ffd24f0 0x400d2fb5:0x3ffd2510 0x400d2482:0x3ffd2530 0x400fc3a6:0x3ffd2550
[10:24:55]
[10:24:55]Rebooting...
[10:24:55]ets Jun  8 2016 00:22:57
[10:24:55]
[10:24:55]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:24:55]configsip: 0, SPIWP:0xee
[10:24:55]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:24:55]mode:DIO, clock div:2
[10:24:55]load:0x3fff0018,len:4
[10:24:55]load:0x3fff001c,len:812
[10:24:55]load:0x40078000,len:0
[10:24:55]load:0x40078000,len:11584
[10:24:55]entry 0x40078a60
[10:24:55][I][log_component:056]: Log initialized
[10:24:55][C][ota:163]: There have been 0 suspected unsuccessful boot attempts.
[10:24:55][I][application:044]: Application::setup()
[10:24:55][C][output.gpio:023]: Setting up GPIO Binary Output...
[10:24:55][C][esphal:053]:     GPIO Pin 17 with mode OUTPUT
[10:24:55][C][output.ledc:026]: Setting up ledc output...
[10:24:55][C][output.ledc:027]:     Pin: 2
[10:24:55][C][output.ledc:028]:     LEDC Channel: 1
[10:24:55][C][output.ledc:029]:     Frequency: 1000.0
[10:24:55][C][output.ledc:030]:     Bit Depth: 12
[10:24:55][C][light.state:179]: Setting up light 'Garage Status LED'...
[10:24:55][C][wifi:031]: Setting up WiFi...
[10:24:55][C][wifi:178]: Setting up STA...
```

I can directly toggle the switch until I'm blue in the face and it works fine, but using the cover will cause a crash every time. I've tried both with Optimistic, and with a Lambda that checks a reed sensor to determine position.

I can provide additional stack traces and the corresponding `firmware.elf` if that helps. If you can point me at a tool to parse the panic message I'd be glad to run that myself.

OttoWinter, 2018-06-04:
Oh noes... I'm using a cover myself, but on an ESP8266 - I'll try to reproduce the issue on my ESP32

So decoding the backtrace *for the ESP32* is relatively simple (in fact, I even wanted to create a helper for esphomeyaml at one point, but now fortunately there are much less panics in esphomelib 😅). 

Note that this is for *nix systems, I have no idea how this would be done on windows.

First, identify the platformio packages folder for the xtensa toolchain. In my case, this is ` ~/.platformio/packages/toolchain-xtensa32/bin`. In that folder, there are a bunch of binaries used for compiling and debugging, for example `xtensa-esp32-elf-g++` and `xtensa-esp32-elf-gdb`.

Now put this path in your global `PATH` variable:

```bash
export PATH=$PATH:/Users/ottowinter/.platformio/packages/toolchain-xtensa32/bin
```

Next, navigate to the platformio project folder. In this folder there should be a file called `platformio.ini`. In this folder, execture `xtensa-esp32-elf-gdb .pioenvs/<NODE_NAME>/firmware.elf`. Now the GDB interactive terminal should open.

Once in there, you basically have access to all gdb commands (and if you're using JTAG, even the step support). But for looking up what's at an address, simply type:
```bash
(gdb) list *0x4014c0d6
...
(gdb) list *0x400d2e7b
...
(gdb) list *0x4014c0da
...
(gdb) list *0x400d2e8b
...
```

You will see the lines surrounding the statement at the given address. Hope it helps :)

brandond, 2018-06-04:
Not sure if I'm doing this wrong, or maybe I need to recompile with some additional options to not strip the binary?

```
user@host:~/repos/hass-esphome$ /home/user/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gdb nodes/garage/.pioenvs/garage/firmware.elf
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from nodes/garage/.pioenvs/garage/firmware.elf...done.
(gdb) list *0x4014c0d6
(gdb) list *0x400d2e7b
(gdb) list *0x4014c0da
(gdb) list *0x400d2e8b
(gdb) list *0x400d5041
(gdb) list *0x400d5131
(gdb) list *0x400d2fb5
(gdb) list *0x400d2482
(gdb) list *0x400fc3a6
```

OttoWinter, 2018-06-04:
Hmm, I could have sworn that that worked before... but I'm seeing the same issue (should have checked before 😅)

A "workaround" is to dump the entire instruction table:

```bash
xtensa-esp32-elf-objdump -x livingroom/.pioenvs/livingroom/firmware.elf -d >symbols.txt
```

In the file `symbols.txt` you will then be able to examine all the code, and at the top of each label there is a comment where that symbol can be found in the source. In the meantime I'll try to get `gdb` working...

brandond, 2018-06-04:
I found this: http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html
Looks like idf_monitor just parses the serial output and calls `xtensa-esp32-elf-addr2line` with the firmware and address. That's easy enough to do:

```
0x4014c0d6: esphomelib::Action<bool>::play_next(bool) at ??:?
0x400d2e7b: esphomelib::switch_::TurnOffAction<bool>::play(bool) at ??:?
0x4014c0da: esphomelib::Action<bool>::play_next(bool) at ??:?
0x400d2e8b: std::_Function_handler<void (), esphomelib::DelayAction<bool>::play(bool)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at ??:?
0x400d5041: esphomelib::Component::loop_internal() at ??:?
0x400d5131: esphomelib::Component::loop_() at ??:?
0x400d2fb5: esphomelib::Application::loop() at ??:?
0x400d2482: loop() at ??:?
0x400fc3a6: loopTask(void*) at ??:?
```

brandond, 2018-06-04:
Even better, I found idf_monitor.py and let it catch a few tests:

```
/home/user/repos/esp-idf/tools/idf_monitor.py --port /dev/ttyUSB0 --toolchain-prefix /home/user/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf- nodes/garage/.pioenvs/garage/firmware.elf
```

They all seem to blow up at the same place:
```
[D][cover.mqtt:045]: 'Garage Door Opener': Closing cover...
[D][binary_sensor.mqtt:066]: 'Garage Door Switch': Sending state ON
[D][cover.mqtt:037]: 'Garage Door Opener': Sending state closed
[D][binary_sensor.mqtt:066]: 'Garage Door Switch': Sending state OFF
[D][binary_sensor.mqtt:066]: 'Garage Door Switch': Sending state ON
[D][binary_sensor.mqtt:066]: 'Garage Door Switch': Sending state OFF
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x4014c0d6  PS      : 0x00060f30  A0      : 0x800d2e7e  A1      : 0x3ffd2450
0x4014c0d6: esphomelib::Action<bool>::play_next(bool) at ??:?

A2      : 0x3ffcdee8  A3      : 0x00000000  A4      : 0x00000050  A5      : 0x3ffd2430
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8014c772  A9      : 0x3ffd2430
A10     : 0xa5a5a5a5  A11     : 0x00000000  A12     : 0x00000005  A13     : 0x00000020
A14     : 0x00000020  A15     : 0x3ffcdf04  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0xa5a5a5a5  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
Backtrace: 0x4014c0d6:0x3ffd2450 0x400d2e7b:0x3ffd2470 0x4014c0da:0x3ffd2490 0x400d2e8b:0x3ffd24b0 0x400d5041:0x3ffd24d0 0x400d5131:0x3ffd24f0 0x400d2fb5:0x3ffd2510 0x400d2482:0x3ffd2530 0x400fc3a6:0x3ffd2550

0x4014c0d6: esphomelib::Action<bool>::play_next(bool) at ??:?
0x400d2e7b: esphomelib::switch_::TurnOffAction<bool>::play(bool) at ??:?
0x4014c0da: esphomelib::Action<bool>::play_next(bool) at ??:?
0x400d2e8b: std::_Function_handler<void (), esphomelib::DelayAction<bool>::play(bool)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at ??:?
0x400d5041: esphomelib::Component::loop_internal() at ??:?
0x400d5131: esphomelib::Component::loop_() at ??:?
0x400d2fb5: esphomelib::Application::loop() at ??:?
0x400d2482: loop() at ??:?
0x400fc3a6: loopTask(void*) at ??:?
```

brandond, 2018-06-04:
Simple fix. I wonder if maybe ESP2866 zeroes all memory, and ESP32 does not?

OttoWinter, 2018-06-04:
Oh thanks very much!

I don't *think* the ESP8266 zeros all memory or that there would be a difference, but who knows.

Maybe one reason might be that your close action is quite long... Mine is quite short:

```yaml
cover:
  - platform: template
    name: "Wohnzimmer Jalousie"
    id: jalousie
    optimistic: True
    open_action:
      - switch.turn_off:
          id: jalousie_runter
      - switch.turn_on:
          id: jalousie_hoch
    close_action:
      - switch.turn_off:
          id: jalousie_hoch
      - switch.turn_on:
          id: jalousie_runter
    stop_action:
      - switch.turn_off:
          id: jalousie_hoch
      - switch.turn_off:
          id: jalousie_runter

binary_sensor:
  - platform: gpio
    pin: D5
    name: "Jalousie Überstrom"
    inverted: True
    id: uberstrom
    on_press:
      then:
        - delay: 100ms
        - lambda:
            if (id(uberstrom).value) {
              id(jalousie).stop();
            }
```

closed 2018-06-04 by OttoWinter

OttoWinter, 2018-06-04:
That esp-idf monitor is just genius! I will try to add it into the native esphomeyaml logs monitor when I get time - the only real issue will be getting the path for the `xtensa32-elf-addr2line` executable from platformio.

commit cc5b6280 (2018-06-04): Fix #55 - Action._next not initialized to nullptr (#57)

Causes play_next() to run off into uninitialized memory on ESP32.