esphome/esphome-core #56: 'System Uptime' sensor filter

brandond, 2018-06-04:
I am using a PIR Motion Sensor. According to [the documentation](https://www.mpja.com/download/31227sc.pdf), any inputs during the first minute of operation should be ignored:

> Induction module needs a minute or so to initialize. During initializing time, it will output 0-3 times. One minute later it comes into standby.

I am not sure exactly what "a minute or so" means, but it would be nice to be able to put an uptime filter on sensors that suppressed readings until the system has been online for a specified period of time.

OttoWinter, 2018-06-04:
I mean I've used those sensors without any initial waiting period for quite a while without issues.

For the uptime filter: Honestly, it feels like quite a limited use case. It would be easy enough to add, but I think esphomelib already probably has too many filters - having this many just confuses people.

If you find another use case, or if you can convince me that this use-case is quite common, please do say so. In the meantime, this should work fine:

`main.cpp`:
```cpp
using namespace esphomelib;

class UptimeFilter : public sensor::Filter {
 public:
  UptimeFilter(uint32_t uptime) : uptime_(uptime) {}
  
  optional<float> new_value(float value) override {
    if (millis() >= uptime_)
      return value;
    return {};
  }
  
  uint32_t uptime_;
};

void setup() {
  // ...
  // =========== AUTO GENERATED CODE END ============
  pir_sensor.add_filter(new UptimeSensor(60*1000));
  App.setup();
}
```

(assuming your PIR sensor is a *sensor*, not a *binary sensor*; filters for binary sensors do not exist *yet*, but will exist soon-ish)

brandond, 2018-06-04:
Yeah, I was just looking at the filter code and realizing that it only worked for full sensors, not binary sensors. Bummer.

I was going to try to plug in some automation that took action on motion detection, and was hoping to avoid as much noise as possible, but I can probably live with it for now.

OttoWinter, 2018-06-04:
Okay. Regarding binary sensor filters: Do you happen to know if there are popular filters apart from the basic "debounce" filter

I was going to make my life easy and just add that one filter (and potentially a "debounce-last" or whatever it's called), but now with this feature request I'm thinking if I should just create a full modular binary sensor filter engine like with the sensors.

brandond, 2018-06-05:
I can't think of much use case for doing a lot of math on binary outputs either; for most of what I want with binaries the filters would just be used to suppress undesired changes to the value.

OttoWinter, 2018-09-27:
Should be really easy now with the uptime sensor. A simple lambda filter should do the trick :)

closed 2018-09-27 by OttoWinter