esphome/esphome-core #58: Uninstall silently esptool !!!

vincegre, 2018-06-05:
Discovered checking what it did when I installed esphomelib and I discovered it removed from my system without asking esptools from it !! What are these gangsters way to do ??
It should ask me if I want to remove it ?? and why does it need to be removed ? conflict in some libs ?

OttoWinter, 2018-06-05:
esphomeyaml doesn't uninstall anything - in fact, it even lists esptool as a requirement for installing esphomeyaml [here](https://github.com/OttoWinter/esphomeyaml/blob/e9ced0cc3f8327362491a39d9b0fa995e789633e/setup.py#L31). So the `pip` installer *should* actually *install* esptool. The installation handling is completely controlled by the `pip` program, as such, esphomeyaml cannot even try to uninstall esptool.

So I think your problem might be related to how you installed esptool previously. I don't know the details of your setup, but I would maybe try seeing if the installation somehow caused a newer version of esptool to be installed, thus potentially *moving* esptool to another location.

See Also: [xkcd python environment](https://m.xkcd.com/1987/)

vincegre, 2018-06-06:
Thanks for explanations and sorry for confusion :-1: I checked more in details the logs of pip while it installed esphomelib and it really removed esptool (existing one, previously installed) but it was to install a newer version :+1: 
Sorry for the wrong alarm !

closed 2018-06-06 by vincegre