esphome/esphome-core #59: ESP8266: Only pins 0-16 support PWM

meekell, 2018-06-05:
I'm attempting to assign a PWM output to GPIO16 of my ESP8266-12E. When attempting to compile, I'm receiving the error: _ERROR [esphomeyaml.config] Error while reading config: ESP8266: Only pins 0-16 support PWM. 

My yaml includes the following:

```
esphomeyaml:
  platform: ESP8266
  board: esp12e

output:
  - platform: esp8266_pwm
    pin: 16
    id: redPin
```

OttoWinter, 2018-06-05:
Oh it seems like the error message is wrong... At least from this source (https://www.esp8266.com/wiki/doku.php?id=esp8266_gpio_pin_allocations) only pins 0-15 support pwm.

I will try to manually check if the source is incorrect and pwm does work on pin 16. If it does work, i'll adjust the validation, otherwise I'll fix the error message

meekell, 2018-06-05:
Thanks for replying. I'm converting my rgbw light controller developed in Arduino IDE to esphomeyaml. In the Arduino code I was using pwm on pin GPIO16 with the analogWrite call, so I know it works.

OttoWinter, 2018-06-05:
Perfect! Thanks!

meekell, 2018-06-05:
Also found 0-16 called out here: [http://esp8266.github.io/Arduino/versions/2.0.0/doc/reference.html](url)

closed 2018-06-05 by OttoWinter

OttoWinter, 2018-06-05:
Thank you! Fixed in the latest dev build now. See https://esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version