esphome/esphome-core #67: Controls internal to device

jellybob, 2018-06-08:
Sorry, I’m not sure how better to word that title, and I’m actually not sure if this is already supported.

I have a Sonoff Touch light switch flashed with esphomelib which is generally working great, however the current setup where pressing a button has to trigger an automation in Home Assistant to toggle the relay has me slightly concerned - if the Home Assistant infrastructure goes down then I’ll have no way to turn the lights on and off.

Ideally I’d like to be able to configure the device itself so that when I press the switch it internally toggles the relay, and then reports that to Home Assistant. More generally this is about devices having a failsafe in situations where network connectivity isn’t available, or if I move house and don’t want to rip out the light switches first.

OttoWinter, 2018-06-08:
So esphomelib/yaml do have automations already: https://esphomelib.com/esphomeyaml/guides/automations.html ... And these do run on the device itself.

Fair warning though: esphomelib currently reboots after a while if no network connection is detected. This is because the WiFi low level stack likes to go into an invalid state every so often and the only way to fix that is with a full restart. I'm working on fixing that though using a more WiFi-independent code base

jellybob, 2018-06-08:
That's exactly what I was looking for, thanks!

closed 2018-06-08 by jellybob