esphome/esphome-core #69: SSD1306 display support

rvanmarkus, 2018-06-08:
Hi, 

Maybe it's a bit redundant and already on the roadmap, but I would really like to see an integration with SSD1306 based OLED displays. I would be interested to help to build this feature, but not sure if I'm experienced enough to write all the code for such a component. 

What for me would be valuable is to have the ability to subscribe to a certain topic for let's say 'notification', or 'current scene' and whenever there is a new valuable published on the topic that is will be displayed on the OLED display. 

What do you think about the first criteria the for the OLED display component? Or what do you guys think about a display component at all? 

Kind regards,
Robbert

brandond, 2018-06-08:
I'd asked for something similar on Discord. I'd love to see it registered as a notification endpoint that I could send things to in Home Assistant for display.

OttoWinter, 2018-06-09:
Yeah displays are definitely something I want to support. Problem is that displays also need a shitton of customizability. I've never really used these sorts of displays before, but I'm now trying to find out what the most common display styles are (I don't want to write an entire rendering engine :) So only the essential parts in the beginning)

OttoWinter, 2018-09-27:
Done :) Is in 1.8.0

closed 2018-09-27 by OttoWinter

rvanmarkus, 2018-09-27:
Awesome, thanks for the work!!

jblb, 2018-09-29:
Hi !
just give a test and nothing is displayed on my OLED :-(
looking forward in .piolibdeps/ directory of my project where is nothing about Adafruit_SSD1306 library.
seem that something is missing somewhere.

OttoWinter, 2018-09-29:
@jblb See https://github.com/OttoWinter/esphomeyaml/issues/128 . There was a bug previously, but it should work with the latest bleeding edge version. Could you check with that?