esphome/esphome-core #72: Output pin `inverted` configuration broken

aequitas, 2018-06-08:
Discovered while trying example from docs: (https://github.com/OttoWinter/esphomedocs/pull/7)

This config is valid, but does not work:
```
output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin: GPIO13
    inverted: true
```

There is no difference in the generated `main.cpp` files if `inverted` is changed from true/false or even is ommited.

Current working configuration:

```
output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin:
      number: GPIO13
      inverted: true
```

brandond, 2018-06-08:
I'd agree that the difference between inverting and output and inverting the pin the output uses can be a bit odd. That said, although all outputs claim to support inversion, only the binary output seems to actually use it. esp8266_pwm defers to output pin inversion, although the source claims otherwise in one spot. See:
https://github.com/OttoWinter/esphomelib/blob/master/src/esphomelib/output/esp8266_pwm_output.h#L39
vs
https://github.com/OttoWinter/esphomelib/blob/master/src/esphomelib/output/esp8266_pwm_output.cpp#L36

OttoWinter, 2018-06-08:
Should be fixed by https://github.com/OttoWinter/esphomeyaml/commit/0be29d27d5064e1f99ce118312db79979137d143. Was a problem with the coroutine-based rewrite.

@brandond That docstring is out of date there and was from the esphomelib-only days. I agree having both on-pin and on-{output,switch,binary_sensor} inversion is weird, but I don't have plans to change that since with the pin schema one can also change the pin **mode**.

brandond, 2018-06-08:
I think your change here may be responsible for this:

```
src/main.cpp: In function 'void setup()':
src/main.cpp:21:87: error: no matching function for call to 'esphomelib::Application::make_monochromatic_light(const char [18], esphomelib::output::GPIOBinaryOutputComponent*&)'
auto application_makelight = App.make_monochromatic_light("Status LED", led_0);
^
In file included from src/main.cpp:2:0:
.piolibdeps/esphomelib/src/esphomelib/application.h:772:13: note: candidate: esphomelib::Application::MakeLight esphomelib::Application::make_monochromatic_light(const string&, esphomelib::output::FloatOutput*)
MakeLight make_monochromatic_light(const std::string &friendly_name, output::FloatOutput *mono);
^
.piolibdeps/esphomelib/src/esphomelib/application.h:772:13: note:   no known conversion for argument 2 from 'esphomelib::output::GPIOBinaryOutputComponent*' to 'esphomelib::output::FloatOutput*'
*** [.pioenvs/cooler/src/main.cpp.o] Error 1
```

closed 2018-06-14 by OttoWinter