esphome/esphome-core #74: [Feature request] Serial input/output

saundersrg, 2018-06-09:
What an awesome project! Particularly keen on the IR side, but coupled with serial. Optoma projectors have all their serial codes published, so controlling a projector via HomeAssistant etc from an ESP would be awesome. TTL to RS232 support would be epic (even just sending commands via ESP's secondary TX would be good).

OttoWinter, 2018-06-12:
*Sending* commands should be relatively straightforward, I will consider it in a future release (not 1.7.0 though, that's pretty much feature-frozen by now).

Receiving would be another issue, but as that's highly dependent on the device used, that will need to be handled on a per-integration basis.

saundersrg, 2018-06-13:
Awesome stuff. For receiving I was thinking more a case of receiving from a projector and pushing to MQTT as states. But can see that receiving could become very device/scenario specific. Happy to help test or anything like that.

notgwj, 2018-06-14:
I have a similar use case with an older Denon Reciever (AVR 3805).  The receiver has a serial port, but no network connection.  I have an esp8266 connected to the serial port and was planning to write code for the esp8266 to translate from the [Denen Serial Protocol](https://usa.denon.com/us/product/hometheater/receivers/avr3808ci?docname=AVR-3808CISerialProtocol_Ver520a.pdf) to MQTT.  I mostly need power / volume control which should be pretty simple.  

If there were a way to do this with esphomeyaml, that would be awesome.

OttoWinter, 2018-06-14:
FWIW, unidirectional serial protocols are pretty much just digital signals like IR remotes or 433MHz signals. So you might be able to use the remote transmitter component to get the UART data as a raw signal and then just send that later. See [this](https://esphomelib.com/esphomeyaml/components/switch/remote_transmitter.html#finding-remote-codes)

saundersrg, 2018-06-15:
Definitely something to try. I'll probably mess around with this on the weekend, thanks! Luckily Optoma publishes all their status messages etc so for my use hopefully it will be relatively straightforward. Thanks for the prompt responses!

OttoWinter, 2018-09-27:
Exists in 1.8.0 using the UART bus component and switch. Closing...

closed 2018-09-27 by OttoWinter