esphome/esphome-core #78: I2CComponent::write_byte_16 writes 0 words

schumar, 2018-06-09:
https://github.com/OttoWinter/esphomelib/blob/249e0b527360e3bc93dcc9b97f698d3f3cb072a9/src/esphomelib/i2c_component.cpp#L196

After changing this to
```
return this->write_bytes_16(address, register_, &data, 1);
```
communicating with a ADS1115 suddenly works :)

OttoWinter, 2018-06-09:
Oh! Thank you so much! That would have been a nightmare to debug (I hope you didn't have to debug for too long... My mistake)

closed 2018-06-09 by OttoWinter

schumar, 2018-06-09:
It wasn't "fun" to debug, in the end I used my Logic Analyzer and hooked it up to see what was going on :)
But don't worry, thanks a lot for esphomelib!

OttoWinter, 2018-06-09:
Oh noes... Sorry about that.

As a tip, you can set the debug level to VERY_VERBOSE and you will see the entire i2c traffic in the logs 😉 https://esphomelib.com/esphomeyaml/components/logger.html

schumar, 2018-06-09:
Sorry for replying to this closed ticket, but I'm trying to debug an issue with TSL2561, and wanted to set the logging level as you suggested:

```
#define ESPHOMELIB_LOG_LEVEL ESPHOMELIB_LOG_LEVEL_VERY_VERBOSE
#include <esphomelib.h>
// [...]
    auto *log = App.init_log(230400);
    log->set_global_log_level(ESPHOMELIB_LOG_LEVEL_VERY_VERBOSE);
```
but I don't get any additional log entries :(

OttoWinter, 2018-06-09:
That would be because `#define` before an include only works with header-only libraries :)

So to set the log level to very verbose you need to set it inside `platformio.ini`:

```ini
[env:YOUR_ENV]
build_flags = -DESPHOMELIB_LOG_LEVEL=ESPHOMELIB_LOG_LEVEL_VERY_VERBOSE
```

schumar, 2018-06-09:
Do you know an easy way to do this using Arduino?

OttoWinter, 2018-06-09:
Puh... that might be difficult. I've never really used the Arduino IDE before, but maybe [this](https://arduino.stackexchange.com/questions/4729/add-d-option-via-boards-txt) could help?

commit 1c9d6ba1 (2018-06-09): Fix I2CComponent::write_byte_16 mistake. Fixes #78