esphome/esphome-core #82: htu21d only works with logger set to very_verbose

ladefoged81, 2018-06-09:
I've been testing the htu21d sensor.
In my test setup it only works with logger set to very_verbose.

If I set it to anything else (default or verbose) it does not works and returns the following:
[23:14:00][I][i2c:051]: Scanning i2c bus for active devices...
[23:14:00][I][i2c:057]: Found i2c device at address 0x40
[23:14:00][E][i2c:125]: Receive timeout for address 0x40
(no values recieved from the sensor)

If logger set to very_verbose it works as expected.

ladefoged81, 2018-07-01:
Any update on this issue?

theres, 2018-07-09:
I've the same issue with PCA9685 on ESP32. The simplest workaround I've found is by setting frequency manually, something like:
```
auto *i2c = App.init_i2c(YOUR_SDA, YOUR_SCL, true); 
i2c->set_frequency(400000);
```

btw. [light example](https://github.com/OttoWinter/esphomelib/blob/master/examples/lights/lights.cpp) has bug - third parameter in `init_i2c` is for 'scan', not for `frequency`

OttoWinter, 2018-09-27:
Can you try with1.8.0? It should be fixed there. Please reopen if not.

closed 2018-09-27 by OttoWinter

ladefoged81, 2018-09-27:
Works like a charm with 1.8.1