esphome/esphome-core #83: Feature request: MQ gas sensor

htcheroportugal, 2018-06-10:
I would like to see any support for the mq gas sensor it was be cool.

OttoWinter, 2018-06-11:
So the MQ-2 gas sensors and similar ones should already be supported by the [ADC Sensor](https://esphomelib.com/esphomeyaml/components/sensor/adc.html). Basically, the MQ gas sensors just output an analog voltage that you need to convert using a formula.

In this case, the formula would be something like `((5.0-x)/x)/9.8`. Look at the TEMT6000 cook book entry for inspiration. I haven't looked up how to hook it up yet, but it shouldn't be too difficult

OttoWinter, 2018-09-27:
Closing as possible with ADC sensor already. Maybe I will write detailed instructions in the cookbook, but I'm closing this issue here as this is the code repo.

closed 2018-09-27 by OttoWinter