esphome/esphome-core #86: [feature-request] use WifiManager-like mechanism as fallback for failing wifi connection

ycardon, 2018-06-11:
Hello, great software here !

Have you considered using the WifiManager lib by tzapu ? 
- https://github.com/tzapu/WiFiManager
- https://platformio.org/lib/show/567/WifiManager

Whenever the wifi connection fails (ie. I'm changing my wifi password or I'm moving to another place), it falls back to an access point providing a little web server to enter the new credentials. It has a quite low memory footprint even for a ESP8266 and it's really convenient for hard-to-flash devices such as Sonoff.

messismore, 2018-06-11:
Related to #36

OttoWinter, 2018-06-12:
I think some of the points of #36 still do apply, but since then I think some stuff also changed:

 * First of all, esphomelib does actually kind of have this functionality already. It's opt-in (you have to have `ap:` in your WiFi configuration), but it exists. The reason why it's opt-in is that I haven't fully evaluated how it impacts the WiFi peripheral and flash size. Especially for the WiFi controller I'm concerned that enabling a WiFi hotspot by default can cause some issues.
 * Even if the WiFiManager does have a low memory footprint, it still does have some. esphomelib believes in the concept that everything that's not completely necessary for operation is disabled unless the user explicitly enables it.
 * The WiFi controller on the ESPs is quite buggy and esphomelib already has to do a bunch of "ugly fixes" to ensure that everything operates correctly. When adding WiFiManager to the equation, there would now be two components controlling the WiFi controller, which would just cause a big mess.

ycardon, 2018-06-13:
Oh sorry I was not aware of the ```ap:``` config. I'm eligible for a RTFM :)

If I understand well, the 'oops I have to fix my wifi' recovery procedure is:
- first insert ```ap:``` along with regular ```wifi:``` in device config
- if the wifi credentials has to be changed: directly connect to the AP and do an OTA from there

Is that it ?

OttoWinter, 2018-09-27:
Yes that should work. Closing as AP failover functionality exists in esphomelib now.

closed 2018-09-27 by OttoWinter