esphome/esphome-core #87: Bosch BME280 reads strange values

timbuchwaldt, 2018-06-11:
I am working on a small sensor node, but during validation of components it showed that data read from a BME280 are very much off what I expected:
Humidity often switches between 100% and 0%, while temperature/pressure seem mostly OK. Sometimes the pressure drops down by like 40hPa (not realistic at all where the sensor lives) or rises to 1224.7hP.

Looking at a scope with I2C decoding turned on it seems like there is valid data coming in for the humidity registers (it also greatly varies when exhaling onto the sensor), but the reported measurements are always 0.0% humidity. Given that the I2C data looks valid we assume some sort of code bug in esphomelib.

OttoWinter, 2018-06-11:
Similar to https://github.com/OttoWinter/esphomeyaml/issues/22 - The fact that this happens randomly leads me to believe that this is probably some sort of timing issue. My BME280 will arrive in a week or two - I will try to fix it then.

timbuchwaldt, 2018-06-11:
Aah damn, didn't search thoroughly enough. What's really strange: I run a ESP8266 for my home setup without any problem (at least I haven't seen one yet) - just my office monitoring setup fails - and is on an ESP32.
I'm just if the wire library is maybe broken (or doesn't use the hardware I2C built into the ESP32)

OttoWinter, 2018-06-12:
AFAIK the Wire library does use the hardware on the ESP32, at least with the current version.

One thing that could be the problem is that esphomelib is waiting too short for the conversion to complete. That could potentially explain the results. I've just multiplied the time to wait for conversion by 1.5 for esphomelib v1.7.0; maybe that can help, at least it shouldn't hurt.

promd, 2018-07-03:
I can also open a different issue, but thought this might be related:
I am using 3 BME280 from 2 different suppliers (Adafruit and noname) on 3 NodeMCU 32S.
Readings for Temperature are always below 1.5° (should be above 20°), readings for humidity are always below 15 % (should be >50%). 
Running on esphomeyaml 1.7.0.

animavitis, 2018-07-04:
@promd: I have same issue with 1.5 temp from bme280 & esp32

mihalski, 2018-07-25:
I'm getting these sort of nonsense values. I know the sensor is good because it is working with Tasmota firmware on the same ESP8266 board (Wemos D1 Mini Pro).
```
[08:23:31][D][sensor.bme280:193]: Got temperature=0.7°C pressure=470807.7hPa humidity=15.1%
```

mihalski, 2018-07-26:
Reverted to Tasmota and BAM. All working fine again:
```
11:45:43 MQT: tele/sonoff/SENSOR = {"Time":"2018-07-26T11:45:43","BME280":{"Temperature":24.3,"Humidity":49.4,"Pressure":1010.5},"BH1750":{"Illuminance":4},"TempUnit":"C"}
```
So it certainly seems like a software issue.

theres, 2018-07-30:
I've very similar results on my site:
> One thing that could be the problem is that esphomelib is waiting too short for the conversion to complete. 

About this:
> One thing that could be the problem is that esphomelib is waiting too short for the conversion to complete. 

I've try to change this to different values - no results. Also, I've experimented with i2c frequrency, measurement mode, standby time -  nothing.

dawsmac, 2018-08-14:
I have also installed one of the BME280 sensors and getting the following results.

D][sensor.bme280:193]: Got temperature=0.2°C pressure=1180357.5hPa humidity=19.5%

If anyone wants me to test any code etc... happy to help out? 

Cheers

Rich

NearlCrews, 2018-08-25:
I've got the same problem. The readings from the Adafruit BME280 are _way_ off when using esphomeyaml. However, if I use the Arduino test sketch with the Adafruit BME280, it'll return the correct results. I'm out of my league on this but would gladly test any potential fixes.

dasmaeh, 2018-09-03:
Same problem for me.
If I create the sensor manually in main.cpp it works though. Does this pull in different librariers then?
Sample code for temperature, leaving out the auto generated stuff:
```
#include "esphomelib/application.h"

#include <Adafruit_BME280.h>
#include <Adafruit_Sensor.h>
#include <Wire.h>

using namespace esphomelib;

class BME280_Temperature : public sensor::PollingSensorComponent {
 public:
   // Load the sensor
   Adafruit_BME280 bme; //(16, 5, 4,  0);

   BME280_Temperature(const std::string &friendly_name, uint32_t update_interval) : sensor::PollingSensorComponent(friendly_name, update_interval) {}
  void setup() override {
    bme.begin();
    // This will be called by App.setup()
  }
  void update() override {
    float temperature = bme.readTemperature();
    // This will be called by App.loop()
    push_new_value(temperature);  // 42°C
  }

  std::string unit_of_measurement() override { return "°C"; }
  int8_t accuracy_decimals() override { return 1; } // 2 decimal places of accuracy.
};
```
```
auto *sensor_temperature = App.register_component(new BME280_Temperature("Terrasse Temperatur", 5000)); // update every 900000ms or every 15 minutes.
  App.register_sensor(sensor_temperature);
```

OttoWinter, 2018-09-26:
Should be fixed in 1.8.0

closed 2018-09-26 by OttoWinter