esphome/esphome-core #88: DHT11 sensor doesn't work with esphomelib

ayavilevich, 2018-06-12:
I have a simple circuit with a DHT11 module connected to an ESP32. Just three wires: 3.3V, Gnd and Pin16 for signal.

When using the following code

```
#include <dht11.h>

dht11 DHT11;
#define DHT11PIN 16

void setup()
{
	Serial.begin(115200);
	while (!Serial); // Wait untilSerial is ready - Leonardo
}

void loop()
{
	int chk = DHT11.read(DHT11PIN);
	Serial.print("Humidity:");// Print a message of "Humidity: ".
	Serial.print((float)DHT11.humidity, 2);
	Serial.print(" %   "); // Print the unit of the humidity.

	Serial.print("Temp:    ");// Print a message of "Temp: ".
	Serial.print((float)DHT11.temperature, 2);// Print a centigrade temperature.
	Serial.println(" C "); // Print the unit of the centigrade temperature.
	delay(1000);
}
```

with the dht11 library

```
//
//    FILE: dht11.cpp
// VERSION: 0.4.1
// PURPOSE: DHT11 Temperature & Humidity Sensor library for Arduino
// LICENSE: GPL v3 (http://www.gnu.org/licenses/gpl.html)
//
// DATASHEET: http://www.micro4you.com/files/sensor/DHT11.pdf
//
// HISTORY:
// George Hadjikyriacou - Original version (??)
// Mod by SimKard - Version 0.2 (24/11/2010)
// Mod by Rob Tillaart - Version 0.3 (28/03/2011)
// + added comments
// + removed all non DHT11 specific code
// + added references
// Mod by Rob Tillaart - Version 0.4 (17/03/2012)
// + added 1.0 support
// Mod by Rob Tillaart - Version 0.4.1 (19/05/2012)
// + added error codes
//

#include "dht11.h"

// Return values:
// DHTLIB_OK
// DHTLIB_ERROR_CHECKSUM
// DHTLIB_ERROR_TIMEOUT
int dht11::read(int pin)
{
	// BUFFER TO RECEIVE
	uint8_t bits[5];
	uint8_t cnt = 7;
	uint8_t idx = 0;

	// EMPTY BUFFER
	for (int i=0; i< 5; i++) bits[i] = 0;

	// REQUEST SAMPLE
	pinMode(pin, OUTPUT);
	digitalWrite(pin, LOW);
	delay(18);
	digitalWrite(pin, HIGH);
	delayMicroseconds(40);
	pinMode(pin, INPUT);

	// ACKNOWLEDGE or TIMEOUT
	unsigned int loopCnt = 10000;
	while(digitalRead(pin) == LOW)
		if (loopCnt-- == 0) return DHTLIB_ERROR_TIMEOUT;

	loopCnt = 10000;
	while(digitalRead(pin) == HIGH)
		if (loopCnt-- == 0) return DHTLIB_ERROR_TIMEOUT;

	// READ OUTPUT - 40 BITS => 5 BYTES or TIMEOUT
	for (int i=0; i<40; i++)
	{
		loopCnt = 10000;
		while(digitalRead(pin) == LOW)
			if (loopCnt-- == 0) return DHTLIB_ERROR_TIMEOUT;

		unsigned long t = micros();

		loopCnt = 10000;
		while(digitalRead(pin) == HIGH)
			if (loopCnt-- == 0) return DHTLIB_ERROR_TIMEOUT;

		if ((micros() - t) > 40) bits[idx] |= (1 << cnt);
		if (cnt == 0)   // next byte?
		{
			cnt = 7;    // restart at MSB
			idx++;      // next byte!
		}
		else cnt--;
	}

	// WRITE TO RIGHT VARS
        // as bits[1] and bits[3] are allways zero they are omitted in formulas.
	humidity    = bits[0]; 
	temperature = bits[2]; 

	uint8_t sum = bits[0] + bits[2];  

	if (bits[4] != sum) return DHTLIB_ERROR_CHECKSUM;
	return DHTLIB_OK;
}
//
// END OF FILE
//
```

I get valid result:

```
Humidity:14.00 %   Temp:    28.00 C 
Humidity:14.00 %   Temp:    28.00 C 
Humidity:14.00 %   Temp:    28.00 C 
Humidity:14.00 %   Temp:    28.00 C 
Humidity:14.00 %   Temp:    28.00 C 
Humidity:14.00 %   Temp:    28.00 C 
Humidity:14.00 %   Temp:    28.00 C 
Humidity:14.00 %   Temp:    28.00 C 
Humidity:14.00 %   Temp:    28.00 C 
```

If using with esphomelib:

```
#include "esphomelib.h"
#define DHT11_PIN 16

using namespace esphomelib;

void setup() {
	App.set_name("esp32");
	App.init_log();

	App.init_wifi("x", "x");
	App.init_ota()->start_safe_mode();
	App.init_mqtt("x", "x", "x");
	App.init_web_server();

	auto dht=App.make_dht_sensor("Office Temperature", "Office Humidity", DHT11_PIN);
	dht.dht->set_dht_model(sensor::DHT_MODEL_DHT11);

	App.setup();
}

void loop() {
	App.loop();
}
```

I get 

```
[0;33m[W][sensor.dht:063]: Invalid readings! Please check your wiring (pull-up resistor, pin number) and consider manually specifying the DHT model using the model option. Error code: TIMEOUT[0m
[0;33m[W][sensor.dht:063]: Invalid readings! Please check your wiring (pull-up resistor, pin number) and consider manually specifying the DHT model using the model option. Error code: TIMEOUT[0m
[0;33m[W][sensor.dht:063]: Invalid readings! Please check your wiring (pull-up resistor, pin number) and consider manually specifying the DHT model using the model option. Error code: TIMEOUT[0m
[0;33m[W][sensor.dht:063]: Invalid readings! Please check your wiring (pull-up resistor, pin number) and consider manually specifying the DHT model using the model option. Error code: TIMEOUT[0m
```

the code in esphomelib seems to be different:

```
uint8_t DHTComponent::read_sensor_(float *temperature, float *humidity) {
  *humidity = NAN;
  *temperature = NAN;

  this->pin_->digital_write(false);
  this->pin_->pin_mode(OUTPUT);

  if (this->model_ == DHT_MODEL_DHT11)
    delay(18);
  else
    delayMicroseconds(800);

  this->pin_->pin_mode(INPUT);
  this->pin_->digital_write(true);

  uint16_t raw_humidity = 0;
  uint16_t raw_temperature = 0;
  uint16_t data = 0;

  for (int8_t i = -3; i < 80; i++) {
    uint8_t age;
    uint32_t start_time = micros();

    do {
      age = micros() - start_time;
      if (age > 90)
        return DHT_ERROR_TIMEOUT;
    } while (this->pin_->digital_read() == ((i & 1) == 1));

    if (i >= 0 && (i & 1) == 1) {
      data <<= 1;

      if (age > 30)
        data |= 1;
    }

    switch (i) {
      case 31:
        raw_humidity = data;
        break;
      case 63:
        raw_temperature = data;
        data = 0;
        break;
      default:break;
    }
  }
```

for example, note 40us delay.

Any ideas?

OttoWinter, 2018-06-13:
Ah yes, something there was wrong. I guess it was the `if (age > 30)` since the DHT11 has a slightly different timing. Anyway: I've reproduced the bug and fixed it at least for my own DHT11 in the latest mater branch (together with some better error reporting).

ayavilevich, 2018-06-14:
@OttoWinter 
got the latest master to try. It wouldn't build due to missing ``esp_err_to_name`` function. So I commented out any use of this function and the dht11 works fine! Seems that ``esp_err_to_name`` is not committed, you might want to take a look at it.
Also, looking at the dht changes you made I tried to fix the original version (just for completeness) and it seems that the cause of the original problem was use of ``delay`` with ``disable_interrupts``. That would not send the proper pulse and the dht11 would not even send anything back. FYI.
Thanks for the fix and for the great library.

OttoWinter, 2018-06-14:
Weird, for me, `esp_err_to_name` is already included by the arduino framework's logging engine, which is included by esphomelib's logging engine. Could you maybe share some information about how you're compiling esphomelib? Using platformio (and the latest arduino framework), it compiles fine for me.

And yes, the `delay` was probably the cause - good catch!