esphome/esphome-core #90: OTA issues due to ".local"

leeharries, 2018-06-12:
I'm using esphomeyaml on RPi 3, and NodeMCU boards. BT Smart Hub with standard firmware.

Uploading via USB works fine for me - I still can't believe how simple it makes this process! Thank you :-)

Trying to upload OTA gives the following error:
ERROR [esphomeyaml.espota] Failed
ERROR [esphomeyaml.espota] Host *board name*.local Not Found

"SHOW LOGS" works fine over the network, so I was curious if this is using a different network name to access the board?

Pinging *board name*.local returns no results. Pinging the board via its IP address works. So it's more than likely a network issue...

_However_, would it be possible to choose to perform OTA via IP address rather than the local network name?

brandond, 2018-06-12:
the somewhat unintuitively named `--upload-port` option should let you specify an IP address for OTA.

leeharries, 2018-06-12:
Thanks @brandond , I'd completely overlooked that in the FAQ, sorry.
Am I right in assuming this can only be achieved via command line? It would be nice to be able submit an IP address in the dashboard.

brandond, 2018-06-12:
The dashboard is still pretty new and doesn't offer a lot of the functionality that you can get from CLI or editing the YAML directly. Guessing not.

OttoWinter, 2018-06-14:
So I don't really want to add a text field into the dashboard, since those are inherently user unfriendly unless done right. In a pre-release version of the dashboard, I had the RUN & UPLOAD dialogs first ask the user where to upload to - with an select+input field and a "SELECT" button.

But as those two actions are very common and pressing a second button just makes the dashboard a lot slower to use, I moved the upload port select into the top bar. 

So to make this work in the dashboard, the upload port selection would need to go into esphomeyaml's configuration and permanently be stored in the YAML. Kind of like an `upload_port` option in the `ota:` section.

However, the problem with that is: what to do if an USB port is plugged in? Would the user need to remove that line again from the configuration then? That would IMO not be a good solution, as editing the configuration should only be required when adding/removing features.

I think a good solution to the problem is quite simple and already exists: the `manual_ip` option in the [`wifi:`](https://esphomelib.com/esphomeyaml/components/wifi.html) section. This will make 1. esphomelib use the provided static IP (and not use DHCP) AND make the OTA process use the provided static IP.

leeharries, 2018-06-14:
Thank you for taking the time to write such a detailed response.

Just tried out setting the manual IP using the wifi section, and it worked like an absolute charm 👍 

Do you want me to close this request?

closed 2018-09-27 by OttoWinter