esphome/esphome-core #91: [Feature Request] Sea level correction for air pressure sensor

exxamalte, 2018-06-13:
It would be nice to be able to configure the altitude in metres for air pressure sensors like the BME280 and use that to calculate the air pressure at sea level.

I just found this code snippet that would help:
```
double bmp_atSealevel(double pressure)
{
  return pressure / pow(1.0 - ((float)altitude / 44330.0), 5.255);
}
```

exxamalte, 2018-06-13:
Workaround for now: Use a filter with `lambda` (or `multiply`), and pre-calculate the right hand side of the above formula based on your individual altitude.
For example, for an altitude of 200 metres, that value is `0.9765179`.
```
  - platform: bme280
    pressure:
      name: Pressure
      filters:
        - lambda: return x / 0.9765179;
```

OttoWinter, 2018-06-13:
This should work too:

```yaml
sensor:
  - platform: bme280
    pressure:
      name: Pressure
      filters:
        - lambda: >-
            return x / powf(1.0 - (<YOUR_ALTITUDE> / 44330.0), 5.255);
```

(`powf` is required here because of C's (poorly designed) math library).

As this is that simple, I don't think this should be an option within esphomeyaml because that would just be a rabbit hole for all sorts of other options. I fear that would in the end make the documentation full of seldomly used features that just clutter up the good stuff.

I think a suitable solution so that others can still find it would be a [cookbook](https://esphomelib.com/esphomeyaml/index.html#cookbook) entry.

exxamalte, 2018-06-13:
Thanks for the alternative suggestion.
I understand your view point. The main reason I was asking for this feature is because I am currently using ESPEasy which lets you define the altitude for the BME280 sensor and does the sea level correction automatically, and I believe that sea level correction is pretty standard when measuring and displaying air pressure.