esphome/esphome-core #93: Presence doppler sensor: RCWL-0516

vincegre, 2018-06-15:
Would be nice to get support for that sensor, very useful for presence detection and lot more efficient than PIR ;)
https://www.14core.com/wiring-the-rcwl0516-auto-induction-doppler-microwave-radar-with-esp826632arduino/
It can be handle as a simple input switch but some anti-rebounce settings would be nice for it !

OttoWinter, 2018-06-15:
Some basic debounce settings are already supported by the binary sensor filters: https://esphomelib.com/esphomeyaml/components/binary_sensor/index.html#binary-sensor-filters

ladefoged81, 2018-06-19:
I use this as a gpio binary_sensor the same way as a PIR sensor.
I've actually use it combined with a PIR sensor and only turn on the lights when both sensors are "on".
Tips: Don't place it to close to the esp board to avoid false positives.

OttoWinter, 2018-09-27:
Closing as it's a simple binary sensor. I have ordered one for myself and wjll uodate the docs with specific instructions once it arrives.

closed 2018-09-27 by OttoWinter