esphome/esphome-core #94: [Feature Request] Please add sensor MH-Z19

nevalain, 2018-06-16:
Co2 sensor MH-Z19 would be great!

balk77, 2018-06-20:
The [ESP Easy plugin](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P049_MHZ19.ino) for MH-Z19 could be used as a basis. Unfortunately I have no clue how to write such module.... I can help debugging though.

OttoWinter, 2018-06-22:
Yep, mine arrived a few days ago. The protocol looks quite simple so I hope I can get it into 1.8.0

ThellraAK, 2018-07-31:
Is UART used anywhere else, I can't find anything searching around, but it seems like it would be easier to add soft UART https://github.com/plieningerweb/esp8266-software-uart and use then be able to just send and receive things to it.

OttoWinter, 2018-09-27:
Done :) it's in 1.8.0

closed 2018-09-27 by OttoWinter