esphome/esphome-core #96: Running platformio failed: [Errno 5] I/O error

romanpeters, 2018-06-17:
I'm trying to flash a Sonoff basic, using the hassio add-on esphomeyaml, with the following config:
```
esphomeyaml:
  name: sonoff
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: '<something>'
  password: '<something>'

mqtt:
  broker: '<something>'
  username: ''
  password: ''

# Enable logging
logger:

ota:
  password: '<something>'
```
The config is valid and it compiles without a problem, but fails every time at uploading. 

```
======================== [SUCCESS] Took 199.89 seconds ========================
INFO [esphomeyaml.__main__] Successfully compiled program.
INFO [esphomeyaml.__main__] Uploading binary...
INFO [esphomeyaml.__main__] Running:  esptool.py --before default_reset --after hard_reset --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /config/esphomeyaml/sonoff/.pioenvs/sonoff/firmware.bin
esptool.py v2.4.0
ERROR [esphomeyaml.__main__] Running platformio failed: [Errno 5] I/O error
ERROR [esphomeyaml.__main__] Please try running esptool.py --before default_reset --after hard_reset --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /config/esphomeyaml/sonoff/.pioenvs/sonoff/firmware.bin locally.
```
I tried removing the compiled `sonoff/` folder, but that didn't help.

OttoWinter, 2018-06-17:
Try unplugging the ESP, restarting the add-on and plugging in the ESP. The I/O error means that no connection to the ESP at `/dev/ttyUSB0` could be made which *could* be because the device was unplugged while the add-on was running,

romanpeters, 2018-06-17:
I think that fixed it. Upload was successful! Thanks 😊

closed 2018-06-17 by romanpeters