esphome/esphome-core #97: [Idea] LEDEffect component

Diaoul, 2018-06-17:
WS2812 (AKA Neopixel) LED strips are very powerful but can be hard to work with as effects have to be implemented on the device and cannot be managed entirely remotely through Home Assistant.
So over time I created an [MQTT Template component](https://github.com/home-assistant/home-assistant/pull/4233), added [Light effects](https://github.com/home-assistant/home-assistant/pull/4538) and more recently worked on my own [LEDEffect library](https://github.com/Diaoul/LEDEffect/) that bridges FastLED+Effects with MQTT+JSON so it can be controlled with Home Assistant using my previous contributions.

You made a great job with esphomelib, I'm eager to try it out however I don't have any "dumb" ESP8266, all have some LED strips attached :sob: 

Does your library currently handle the case for programmable LED strips? If not, do you have some plans for it? I'd be happy to discuss some level of integration of the LEDEffect library with esphomelib, I think it would be a very neat integration.

OttoWinter, 2018-06-18:
So there is already the FastLED light component: https://esphomelib.com/esphomeyaml/index.html#light-components which handles these individually addressable LED strips.

But that integration doesn't have too many light effects yet. I'm not a big fan of modifying the JSON payload for the MQTT JSON light though, as that would require special automations in the Hass configuration. I'm planning to rework the effects in esphomelib a bit to allow at least for some parameterization of settings. The syntax would look like this:

```yaml
light:
  - platform: fastled_clockless
    # ...
    effects:
      - rainbow:
          name: "Slow Rainbow Effect"
          speed: 42s # or whatever this option would be called
      - rainbow:
          name: "Fast Rainbow Effect"
          speed: 1s
      # or when no parameters specified defaults to some values
      - rainbow:
      - random:
      - colorwheel:
```

This way the user has complete controls of which effects can be used and can even parameterize them quite quickly without messing with sending custom MQTT messages in HA.

Diaoul, 2018-06-20:
This is the kind of integration that I had in mind.

I'm not a big fan of using a custom JSON payload either because of the automations required you mention however there was some limitations with the others that I didn't like. I can't recall all the reasons but one is that RGB is hardcoded and you can set both RGB and an effect which makes absolutely no sense. MQTT JSON was designed for single color led strips, not for individually adressable LEDs. I don't know how you worked around that but I was not happy with it.

At first I wanted to allow customization on the effect level which is currently not supported by Home Assistant. e.g. change the speed of the Rainbow while staying on the effect "Rainbow".
Then I went back and simplified it for easier daily use. e.g. use two effects "Fast Rainbow" and "Slow Rainbow" with fixed speeds. I think this is the best way to do it.

How is the data transmitted to the ESP currently? Using MQTT JSON Light? Or just MQTT Light? Or something else? I wanted the ESP to expose a REST API and not only work with Home Assistant, I like interoperability.

Also, the possibility to program the ESP using yaml is brilliant I must say. Very easy to manage without hopefully bloating too much the configuration file with ESP-side stuff.

OttoWinter, 2018-06-22:
> I don't know how you worked around that but I was not happy with 

Well I created the first version (not the one you see today) of the MQTT JSON light platform in hass so... :) Home Assistant's architecture is built for single lights, as having support for individually addressable lights *in hass* would be a nightmare API-wise.

There's been some talk recently about changing the effects API in Hass, but that will take a long time to complete (if at all).

> How is the data transmitted to the ESP currently?

Via the MQTT JSON light platform.

> I wanted the ESP to expose a REST API and not only work with Home Assistant, I like interoperability.

Yeah I do too. The thing is that esphomelib is made in such a way that the frontend (aka MQTT JSON or REST API) should have no knowledge of what sits in the back end (like neopixel, simple RGB or whatever). So there would need to be an internal API between effects options and the front end.

bthome, 2018-07-14:
Here are a few project inspirations...

https://github.com/toblum/McLighting  (has extensive effects library)

https://github.com/DedeHai/FastLEDPainter
https://github.com/DedeHai/NeoPixelPainter

OttoWinter, 2018-09-27:
The effects engine now is in esphomelib 1.8.0

closed 2018-09-27 by OttoWinter

Diaoul, 2018-09-27:
Cool ! I'll give it a shot.

Diaoul, 2019-04-21:
7 months later, I tried it but here is my first hurdle: https://github.com/esphome/issues/issues/253