esphome/esphome-core #100: After soft reset or HASS restart, DHT sensor doesn't register

mastershake8241, 2018-06-20:
Setup is a nodemcu v2, with a DHT22 powered by the 3.3v pin. It has a builtin resistor on it's little board it came with. The problem is after a soft reset by pushing the nodemcu's button to reset it, or a restart of HASS it doesn't see the DHT22 sensor is connected anymore unless I physically disconnect and reconnect the sensor or pull it's power cord to the nodemcu and reconnect it. I see this output when it's not working and it will output this until 1 of the 2 things happens above. I let it go for a couple hours to see if it would fix itself.

`[08:19:39][W][sensor.dht:118]: Falling edge for bit -1 failed!
[08:19:39][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0`

I also wrote a basic DHT22 firmware for it and uploaded to test if it was the sensors or the nodemcu and with just the generic DHT esp8266 library it has no problem continuing to see the sensor after multiple soft resets and outputting the temp/humidity.

My config file is:
```
esphomeyaml:
  name: kitchen
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: '********'
  password: '********'

mqtt:
  broker: '192.168.1.121'
  username: ''
  password: ''

# Enable logging
logger:

ota:

sensor:
  - platform: dht
    pin: D4
    temperature:
      name: "Kitchen Temp"
    humidity:
      name: "Kitchen Humidity"
    model: DHT22
```

brandond, 2018-06-20:
I wonder if maybe the 3.3v voltage is dropping too low on soft reset with wifi and everything turned on. How long is the cable? Are you using 3.3v from the onboard USB 5v to 3.3v VRM, or are you supplying external power?

I found that I had to run mine off 5v in order to get them to function reliably. Note that this requires adding an additional resistor between pin 2 and ground in order to take the voltage down to 3.3v in order to avoid frying the inputs on the ESP.

mastershake8241, 2018-06-20:
I apologize, I had it on the 5V pin supplied by the input voltage of the USB cable/power supply. I tried it on the 3.3V and it's the same result. Hmm I guess I'll try adding a resistor. When you say pin 2 do you mean D2 or GPIO 2. The DHT22 comes with a resistor built onto it's little board. I believe it says 512 on the smd resistor which is 5.1 kΩ rated.

brandond, 2018-06-20:
I found some notes here on the wiring: https://raspberrypi.stackexchange.com/questions/27144/how-to-hook-up-an-am2302-temperature-humidity-sensor-several-feet-away-from-th#comment33987_27146

I too have a breakout board with a built-in 5.1k resistor between data and vcc, so I only had to add another 7.6k resistor between data and ground to get it down into a range that won't fry the GPIO input with 5v logic. It's just a basic voltage divider circuit.

mastershake8241, 2018-06-21:
I forgot to mention the cable length is 4 inches. I'll try the extra resistor and report results.

mastershake8241, 2018-06-21:
So I tried using a 4.7K and 10K resistor between the data and ground but had no luck. I don't have any 7.6K resistors. Then I did more research and it's apparently a common problem with eps8266's and DHT22's. The theory I saw most was that the wifi of the esp kind of blocks the sensor in some way until it's replugged in. I saw mention that using a decoupling capacitor was the solution but I only have 10uF and 3.3uF or higher caps. It was suggested to use a .1uf and then a 10uF cap on the vcc and gnd wires. I tried it with just a 10uF cap, just a 3.3uF cap and then both but didn't do anything different. 

I noticed though if I set one of the other pins as a switch and connected the GND wire of the DHT22 to it then during bootup it must reset it maybe but the Sensor worked right off the bat at boot with no other resistors than the 5.1K one on the breakout board of the sensor. Maybe not the best solution and there's a button in HASS now that basically is for nothing really. I honestly don't have a lot of time troubleshoot this so for now it's a solution that seems to work.

brandond, 2018-06-21:
Interesting - the pins do get pulled up or down in interesting ways on boot. I have mine attached long-term to an ESP32, it's possible I just didn't run into that while prototyping with the ESP8266. 

With recent versions of esphomeyaml, if you don't give the switch a name (just an ID), it won't register with MQTT and won't show up in HASS. Might be an acceptable workaround.

closed 2018-06-24 by mastershake8241

yottatsa, 2019-02-07:
Having the same with ESP-01 and DHT22. The wiring is like this (ignore DHT11, I've re-soldered it)
[image]

After OTA or soft-reset or reset via button I'm getting the same error (it's now looks different):
```
[W][sensor.dht:145]: Requesting data from DHT failed!
[W][sensor.dht:079]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
```

The only way to revive it is to powercycle ESP.

remlabm, 2019-02-08:
i can report, i'm having the same issues with my ESP32 and D1 Mini's.. 

only a power cycle fixes the issue..