esphome/esphome-core #104: [Feature Request] RF Receiver / Transmitter

martinjuhasz, 2018-06-29:
Hello,

i would love to be able to use an rf receiver and transmitter. Is there any plan on integrating such?

OttoWinter, 2018-07-11:
Yes, those are already supported here: https://esphomelib.com/esphomeyaml/index.html#misc-components (see "remote receiver" and "remote transmitter") - currently, you need to "sniff" the codes from your RF remote manually, but there's some work to make that a bit easier.

closed 2018-09-27 by OttoWinter

HFsi, 2019-02-07:
Which one did you use or recommend? I mean like the arduino modules....DHT11, ADS1115 and so on.
Is there anyone you could say is the best for ESPHome?

OttoWinter, 2019-02-07:
@HFsi I do not use those myself (except for RF transmitters/receivers with a proprietary format; and non-433MHz frequency)