esphome/esphome-core #105: [Feature Request] Support multiple FastLED controller

thoscut, 2018-07-03:
Currently only one controller is supported, see https://github.com/OttoWinter/esphomelib/blob/4abcccd0ed7de77f23a31c5e01edce8470a727dd/src/esphomelib/light/fast_led_light_output.cpp#L87
It would be great to allow controlling multiple, independently connected leds.

OttoWinter, 2018-07-11:
Well you can control multiple FastLED lights without a problem. Just create two light components in esphomelib/yaml and it should work. The line you're referring to there is just there to protect from a single light *component* having two fastled strips attached to it.

thoscut, 2018-09-02:
Hi Otto,

Below my setup in the yaml. When I comment one of the fastled components, no matter which one, it works flawlessly, but with both enabled, it repeatedly crashes during setup. Platform is an ESP32 (esp32doit-devkit-v1).

YAML:
`
output:
  - platform: ledc
    pin: 18
    id: output_component1
    
light:
  - platform: monochromatic
    name: "Floor Light"
    output: output_component1
  - platform: fastled_clockless
    chipset: WS2812
    pin: GPIO14
    num_leds: 16
    rgb_order: GRB
    name: "Floor Ledring"
  - platform: fastled_clockless
    chipset: WS2811
    pin: GPIO13
    num_leds: 1
    rgb_order: GRB
    name: "Floor Ledpoint"
`

Log:
`
[20:52:41][I][log_component:056]: Log initialized
[20:52:41][C][ota:182]: There have been 5 suspected unsuccessful boot attempts.
[20:52:41][I][application:048]: Running through setup()...
[20:52:41][C][output.ledc:026]: Setting up ledc output...
[20:52:41][C][output.ledc:027]:     Pin: 18
[20:52:41][C][output.ledc:028]:     LEDC Channel: 1
[20:52:41][C][output.ledc:029]:     Frequency: 1000.0
[20:52:41][C][output.ledc:030]:     Bit Depth: 12
[20:52:41][C][light.fast_led:037]: Setting up FastLED light...
[20:52:41][C][light.fast_led:044]:     Max refresh rate: 400
[20:52:41][C][light.fast_led:037]: Setting up FastLED light...
[20:52:41][C][light.fast_led:044]:     Max refresh rate: 400
[20:52:41][C][light.state:180]: Setting up light 'Floor Light'...
[20:52:41][C][light.state:180]: Setting up light 'Floor Ledpoint'...
[20:52:41][C][light.state:180]: Setting up light 'Floor Ledring'...
[20:52:41][C][wifi:031]: Setting up WiFi...
[20:52:41][I][wifi:227]: WiFi Connecting to 'mywifinetwork'...
[20:52:41][C][wifi:244]:     STA Hostname: 'ts_floorlight'
[20:52:41][W][wifi:326]: WiFi Unknown connection status 255
[20:52:41]abort() was called at PC 0x4008445d on core 1
[20:52:41]
[20:52:41]Backtrace: 0x4008e304:0x3ffc0770 0x4008e4df:0x3ffc0790 0x4008445d:0x3ffc07b0 0x4008457d:0x3ffc07e0 0x400edc5b:0x3ffc0800 0x400e9ad5:0x3ffc0ac0 0x400e2fb0:0x3ffc0b10 0x400d2969:0x3ffc0b40 0x400d2b0a:0x3ffc0b60 0x400d313b:0x3ffc0b90 0x400d319a:0x3ffc0bb0 0x40081455:0x3ffc0be0 0x40081991:0x3ffc0c10 0x400d291a:0x00000000
[20:52:41]
[20:52:41]Rebooting...
[20:52:41]ets Jun  8 2016 00:22:57
[20:52:41]
[20:52:42]rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[20:52:42]configsip: 0, SPIWP:0xee
[20:52:42]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:52:42]mode:DIO, clock div:2
[20:52:42]load:0x3fff0018,len:4
[20:52:42]load:0x3fff001c,len:812
[20:52:42]load:0x40078000,len:0
[20:52:42]load:0x40078000,len:11584
[20:52:42]entry 0x40078a60
[20:52:42][I][log_component:056]: Log initialized
[20:52:42][C][ota:182]: There have been 6 suspected unsuccessful boot attempts.
[20:52:42][I][application:048]: Running through setup()...
[20:52:42][C][output.ledc:026]: Setting up ledc output...
[20:52:42][C][output.ledc:027]:     Pin: 18
[20:52:42][C][output.ledc:028]:     LEDC Channel: 1
[20:52:42][C][output.ledc:029]:     Frequency: 1000.0
[20:52:42][C][output.ledc:030]:     Bit Depth: 12
[20:52:42][C][light.fast_led:037]: Setting up FastLED light...
[20:52:42][C][light.fast_led:044]:     Max refresh rate: 400
[20:52:42][C][light.fast_led:037]: Setting up FastLED light...
[20:52:42][C][light.fast_led:044]:     Max refresh rate: 400
[20:52:42][C][light.state:180]: Setting up light 'Floor Light'...
[20:52:42][C][light.state:180]: Setting up light 'Floor Ledpoint'...
[20:52:42][C][light.state:180]: Setting up light 'Floor Ledring'...
[20:52:42][C][wifi:031]: Setting up WiFi...
[20:52:42][I][wifi:227]: WiFi Connecting to 'mywifinetwork'...
[20:52:42][C][wifi:244]:     STA Hostname: 'ts_floorlight'
[20:52:42][W][wifi:326]: WiFi Unknown connection status 255
[20:52:42]abort() was called at PC 0x4008445d on core 1
[20:52:42]
[20:52:42]Backtrace: 0x4008e304:0x3ffc0770 0x4008e4df:0x3ffc0790 0x4008445d:0x3ffc07b0 0x4008457d:0x3ffc07e0 0x400edc5b:0x3ffc0800 0x400e9ad5:0x3ffc0ac0 0x400e2fb0:0x3ffc0b10 0x400d2969:0x3ffc0b40 0x400d2b0a:0x3ffc0b60 0x400d313b:0x3ffc0b90 0x400d319a:0x3ffc0bb0 0x40081455:0x3ffc0be0 0x40081991:0x3ffc0c10 0x400d291a:0x00000000
[20:52:42]
[20:52:42]Rebooting...
[20:52:42]ets Jun  8 2016 00:22:57
[20:52:42]
[20:52:42]rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[20:52:42]configsip: 0, SPIWP:0xee
[20:52:42]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:52:42]mode:DIO, clock div:2
[20:52:42]load:0x3fff0018,len:4
[20:52:42]load:0x3fff001c,len:812
[20:52:42]load:0x40078000,len:0
[20:52:43]load:0x40078000,len:11584
[20:52:43]entry 0x40078a60
[20:52:43][I][log_component:056]: Log initialized
[20:52:43][C][ota:182]: There have been 7 suspected unsuccessful boot attempts.
[20:52:43][I][application:048]: Running through setup()...
`

OttoWinter, 2018-09-20:
Ah ok, now I understand. That was a problem with the fastled library but should be fixed with the next esphomelib release in a few days.

closed 2018-09-20 by OttoWinter

thoscut, 2018-09-26:
Can confirm it's working now.