esphome/esphome-core #110: [Feature Request] Support for current sensors?

derandiunddasbo, 2018-07-24:
I'd like to do some automations in HA depending on current flow. F.e. switching on the cooker hood, if the stove is running.

Are there already any AC current sensors supported by esphomelib?

The non invasive Sodial SCT 013-030 current sensors seem to be quite common and there are already good arduino implementations for it:

https://learn.openenergymonitor.org/electricity-monitoring/ct-sensors/interface-with-arduino

Is there a chance to have these sensors supported by esphomelib?

OttoWinter, 2018-07-27:
I have recently bought some INA3221 and INA219 modules, but those are primarily made for measuring DC current. Maybe these SCT 013-030 sensors will be supported at some point by esphomelib, but I think currently the best way to go about that is using a custom sensor (https://esphomelib.com/esphomeyaml/components/sensor/custom.html - tutorial currently broken, but I'm working on fixing that) with the code you linked to earlier.

OttoWinter, 2018-11-27:
A lot of those sensors have now been implemented

closed 2018-11-27 by OttoWinter