esphome/esphome-core #112: Compile Errors

CrafterLaughter, 2018-07-26:
Using primary sketch example Otto has provided on the esphomelib github (internal PWM and a DHT22) I'm running into compile errors with Arduino 1.8.5 for the custom JSON library (bufferbase?).  I have all 7 of the noted depency libraries loaded.  I'm using the portable directory option Arduino IDE supports.  Any guidance that can be provided as to what might be going on is greatly appreciated.  Thanks!

Here's the compile output starting from it's error line...

`In file included from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/JsonBufferBase.hpp:7:0,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/DynamicJsonBuffer.hpp:7,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson.hpp:7,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson.h:9,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\esphomelib-1.6.0\src/esphomelib/helpers.h:13,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\esphomelib-1.6.0\src\esphomelib\helpers.cpp:14:

C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/Deserialization/JsonParser.hpp: In instantiation of 'struct ArduinoJson::Internals::JsonParserBuilder<ArduinoJson::Internals::StaticJsonBufferBase, const std::basic_string<char>, void>':

C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/Deserialization/JsonParser.hpp:96:66:   required by substitution of 'template<class TJsonBuffer, class TString> typename ArduinoJson::Internals::JsonParserBuilder<TJsonBuffer, TString>::TParser ArduinoJson::Internals::makeParser(TJsonBuffer*, TString&, uint8_t) [with TJsonBuffer = ArduinoJson::Internals::StaticJsonBufferBase; TString = const std::basic_string<char>]'

C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/JsonBufferBase.hpp:70:60:   required from 'typename ArduinoJson::Internals::EnableIf<(! ArduinoJson::Internals::IsArray<TString>::value), ArduinoJson::JsonObject&>::type ArduinoJson::Internals::JsonBufferBase<TDerived>::parseObject(const TString&, uint8_t) [with TString = std::basic_string<char>; TDerived = ArduinoJson::Internals::StaticJsonBufferBase; typename ArduinoJson::Internals::EnableIf<(! ArduinoJson::Internals::IsArray<TString>::value), ArduinoJson::JsonObject&>::type = ArduinoJson::JsonObject&; uint8_t = unsigned char]'

C:\arduino-1.8.5-windows\portable\sketchbook\libraries\esphomelib-1.6.0\src\esphomelib\helpers.cpp:144:45:   required from here

C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/Deserialization/JsonParser.hpp:73:50: error: no type named 'Reader' in 'struct ArduinoJson::Internals::StringTraits<const std::basic_string<char>, void>'

   typedef typename StringTraits<TString>::Reader InputReader;

                                                  ^

C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/Deserialization/JsonParser.hpp:74:50: error: no type named 'Reader' in 'struct ArduinoJson::Internals::StringTraits<const std::basic_string<char>, void>'

   typedef JsonParser<InputReader, TJsonBuffer &> TParser;

                                                  ^

In file included from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/DynamicJsonBuffer.hpp:7:0,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson.hpp:7,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson.h:9,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\esphomelib-1.6.0\src/esphomelib/helpers.h:13,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\esphomelib-1.6.0\src\esphomelib\helpers.cpp:14:

C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/JsonBufferBase.hpp: In instantiation of 'typename ArduinoJson::Internals::EnableIf<(! ArduinoJson::Internals::IsArray<TString>::value), ArduinoJson::JsonObject&>::type ArduinoJson::Internals::JsonBufferBase<TDerived>::parseObject(const TString&, uint8_t) [with TString = std::basic_string<char>; TDerived = ArduinoJson::Internals::StaticJsonBufferBase; typename ArduinoJson::Internals::EnableIf<(! ArduinoJson::Internals::IsArray<TString>::value), ArduinoJson::JsonObject&>::type = ArduinoJson::JsonObject&; uint8_t = unsigned char]':

C:\arduino-1.8.5-windows\portable\sketchbook\libraries\esphomelib-1.6.0\src\esphomelib\helpers.cpp:144:45:   required from here

C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/JsonBufferBase.hpp:70:60: error: no matching function for call to 'makeParser(ArduinoJson::Internals::StaticJsonBufferBase*, const std::basic_string<char>&, uint8_t&)'

     return Internals::makeParser(that(), json, nestingLimit).parseObject();

                                                            ^

C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/JsonBufferBase.hpp:70:60: note: candidate is:

In file included from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/JsonBufferBase.hpp:7:0,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/DynamicJsonBuffer.hpp:7,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson.hpp:7,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson.h:9,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\esphomelib-1.6.0\src/esphomelib/helpers.h:13,

                 from C:\arduino-1.8.5-windows\portable\sketchbook\libraries\esphomelib-1.6.0\src\esphomelib\helpers.cpp:14:

[... 58 lines cut ...]
C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2\src/ArduinoJson/Deserialization/../Serialization/JsonWriter.hpp:141:13: error: 'class std::basic_string<char>' has no member named 'print'

     _length += _sink.print(s);

             ^

Multiple libraries were found for "WiFiClient.h"
 Used: C:\arduino-1.8.5-windows\portable\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi
 Not used: C:\arduino-1.8.5-windows\libraries\WiFi
 Not used: C:\arduino-1.8.5-windows\libraries\WiFi
 Not used: C:\arduino-1.8.5-windows\libraries\WiFi
 Not used: C:\arduino-1.8.5-windows\libraries\WiFi
Using library esphomelib-1.6.0 at version 1.6.0 in folder: C:\arduino-1.8.5-windows\portable\sketchbook\libraries\esphomelib-1.6.0 
Using library ArduinoJson-5.13.2 at version 5.13.2 in folder: C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ArduinoJson-5.13.2 
Using library async-mqtt-client-master at version 0.8.2 in folder: C:\arduino-1.8.5-windows\portable\sketchbook\libraries\async-mqtt-client-master 
Using library ESPAsyncTCP-master at version 1.2.0 in folder: C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ESPAsyncTCP-master 
Using library ESP8266WiFi at version 1.0 in folder: C:\arduino-1.8.5-windows\portable\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi 
Using library Wire at version 1.0 in folder: C:\arduino-1.8.5-windows\portable\packages\esp8266\hardware\esp8266\2.4.1\libraries\Wire 
Using library FastLED-3.1.8 at version 3.1.6 in folder: C:\arduino-1.8.5-windows\portable\sketchbook\libraries\FastLED-3.1.8 
Using library ESPAsyncWebServer-master at version 1.2.0 in folder: C:\arduino-1.8.5-windows\portable\sketchbook\libraries\ESPAsyncWebServer-master 
Using library ArduinoOTA at version 1.0 in folder: C:\arduino-1.8.5-windows\portable\packages\esp8266\hardware\esp8266\2.4.1\libraries\ArduinoOTA 
Using library ESP8266mDNS in folder: C:\arduino-1.8.5-windows\portable\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266mDNS (legacy)
Using library Hash at version 1.0 in folder: C:\arduino-1.8.5-windows\portable\packages\esp8266\hardware\esp8266\2.4.1\libraries\Hash 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).`

OttoWinter, 2018-07-27:
Are you sure you installed the custom ArduinoJSON library I created? https://github.com/OttoWinter/ArduinoJson/releases

The public ArduinoJSON library (https://github.com/bblanchon/ArduinoJson) is not compatible with esphomelib. Also, I would recommend using platformio with esphomelib - especially when it comes to using external libraries it's just much easier to use.

CrafterLaughter, 2018-07-27:
Yes.  The ArduinoJson-5.13.2 noted in the compile output is your custom one.  I've never used the public 5.13.2 before.  I'll give platformio a shot.  Thanks and have a great weekend.

OttoWinter, 2018-11-27:
Closing as this seems to be an external issue with PlatformIO (and this error should be tracked in the documentation repo if it still applies).

closed 2018-11-27 by OttoWinter