esphome/esphome-core #113: wifi connection problems

tomeeka, 2018-07-28:
Hello Otto,

I am using wemos d1 mini boards with esp easy, but i want to migrate to esphome. (I hate clicking on that slow webconfig menu...)
I have tried esphomelib&yaml, and they are are great!
But I have a few wifi issues...

1.
Cant connect to hidden ssid.

2.
I have a wemos d1 mini in a room where the wifi signal is weaker. With esphomelib, it does not connect to my router. If I move the board closer, than it is ok. With esp easy I have no such issue before. I compiled the sample webserver project with platform io and it is working fine too in that room. (restarted the board several times, and in a few seconds it reconnects, but not with esphome)

I plugged the board into my laptop and checked the logs through serial.
[image]

One time the project crashed and never recovered from that point.
[image]

Could you help me with these? I really like esphomelib&yaml but I have 2 more boards that are far from the router, and I am affraid that if I upload new firmware on them they will never come back.

Thanks!

tomeeka, 2018-07-30:
Tested with a different board (maybe hw problem) but same issue. :(

TheJulianJES, 2018-07-31:
I flashed esphomelib with esphomeyaml on some NodeMCU boads (worked perfectly) and also on three Sonoff Basics, only two of them worked. And the other one didn't connect to the wifi at all and rebooted. I'll try to flash Tasmota and see if that connects to WiFi. If that works, I'll try flashing esphomelib again.

ormstont, 2018-08-01:
I'm having a very similar issue - tried it on several Wemos D1 Minis and had no problems at all, connected straight away. I tried it on two Sonoff Basics and one worked initially but then after a few more flashes wouldn't connect. The other never connected. Tasmota on them connects with no problems.

stoinov, 2018-08-18:
Same here with latest 1.7.0 - Wemos D1 mini unable to connect even though tasmota works on it.

aptonline, 2018-08-29:
Yep getting the same here on both Wemos D1 mini and a NodeMCU, both work fine with Tasmota. I may have wrongly posted this issue in the esphomeyaml issues:

https://github.com/OttoWinter/esphomeyaml/issues/108

jollybully, 2018-08-30:
@inforsgit I'm getting the exact same issue, initial upload works with sonoff basic, subsequent uploads knock out WiFi. WiFi works fine with Tasmoata

AlexMekkering, 2018-09-02:
I'm experiencing the same issues with a Wemos D1 Mini pro and a NodeMCU v2. I've tried two different Wifi networks and they simply won't connect.
An ESP32 I have connects without problems.

OttoWinter, 2018-09-25:
esphomelib/yaml v1.8.0 now defaults to choosing a relatively recent arduino framework version which works a lot better than the previous default. This, I hope at least, should eliminate at least part of this problem.

tomeeka, 2018-10-05:
Hi Otto, I have been testing the new version since last night. So far so good. :) I really appreciate your hard work! Greetings from Hungary!

OttoWinter, 2018-11-27:
Closing this issue as it has apparently gotten a lot better in recent versions from what I've heard so far. It's still not perfect but for further discussion I will set up a central issue where all these "wifi and/or mqtt" issues will be linked to.

closed 2018-11-27 by OttoWinter